ArduPilot/ardupilot #29612: Using QMC5883P will not be able to complete Compass Calibration

Scarittagle, 2025-03-25:
## Bug report

**Issue details**

Using QMC5883P will not be able to complete compass calibration, no matter how I reboot or choose relaxed input, it never works. However, the older model QMC5883L works very well. I think this is some kind of compatibility issue.

I recently received an update from the manufacturer states that **they will be discontinue QMC5883L and their last shipment will be on June 20th, 2025** and encourage to use QMC5883P as replacement. It is important for our(and many others) boards which currently using QMC5883L as the mag sensor can continue to ship out hardware without undergo revision and retest for a new mag sensor.

**Version**
4.6

**Platform**
[  ] All
[ x ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
No airframe, tested on bare board

**Hardware type**
Customed Board (STM32F407)

**Logs**
No Logs.

andyp1per, 2025-03-25:
Are you able to send me a board with this device?

andyp1per, 2025-03-25:
> > Are you able to send me a board with this device?
> 
> Sure, not a problem. Can you shoot me an email with the address?

Need to know your e-mail address to do that

Scarittagle, 2025-03-25:
> > > Are you able to send me a board with this device?
> > 
> > 
> > Sure, not a problem. Can you shoot me an email with the address?
> 
> Need to know your e-mail address to do that

suw@sigsa-engineering.com

peterbarker, 2025-05-01:
@Scarittagle did the board get sent?

@andyp1per did the board arrive?

peterbarker, 2025-05-01:
Note that ArduPilot should already support the 5883P.

However, there's *also* this issue: https://github.com/ArduPilot/ardupilot/pull/27156 - it's possible that the 5883P suffers from the same problem as the L - same part, different scaling (TBC).

Scarittagle, 2025-05-05:
@andyp1per Does calibrate with 5883P alone works out for you?

andyp1per, 2025-05-07:
Works fine for me.

Example here:
https://github.com/andyp1per/ardupilot/tree/pr-qmc5883p

andyp1per, 2025-05-10:
Yes, working now

closed 2025-05-10 by andyp1per