ArduPilot/ardupilot #29630: AP_Proximity_Rangefinder: Repoorting out of range data as obstacles

horiuchi0photo, 2025-03-27:
## Bug report

### My configuration
I'm using rangefinder connected to CANNode as a proximity sensor.
```mermaid
graph LR
FC[CubeOrange] <-->|CAN| CANNode[MatekL431<br>AP_Periph]<-->|Serial|RF[Lightware LW20/C]
```

When the range finder detects an object, it reports it as an obstacle (normal).
Then when the range finder lost the object, it reports it as "OutOfRange" (RFND.stat = 2 or 3). But the proximity sensor keeps reporting the last seen obstacle.

[image]

**Version**
Copter-4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
CubeOrange
MatekL431 (AP_Periph-1.7.0 MatekL431-Rangefinder)
Lightware LW20/C

**Logs**
[DataFlash log](https://drive.google.com/file/d/169wHvt_wpShQbWbweMFhHThZ1-PgS1lK/view?usp=sharing)

rmackay9, 2025-03-27:
Hi @horiuchi0photo,

The proximity library has some filtering that will make the obstacles continue to appear for a bit but they should disappear within a couple of seconds.  Maybe this is what you're seeing?  The amount of filtering is configurable I think.

horiuchi0photo, 2025-03-27:
The obstacle indicators did not disappear for a long time.
As the log shows, the rangefinder distance is reported regardless of RFND.stat, and AP_Proximity_Rangefinder uses that distance, so the obstacle reports do not disappear.