ArduPilot/ardupilot #29638: The Repository is feeling bloated. A lot of RPC Failed and early EOF error while cloning even from the fork

sanggusti, 2025-03-28:
## Bug report

When cloning the repository, I've felt there is a lot of early EOF, unexpected Disconnect and RPC failed which usually happens when the repository is too large. As I see in the repository, there are a lot of stale branches (71 branches with updates only on 6 build branches). I had to resort on cloning the repository into several adjustments such as:
- Turn off compression (`git config --global core.compression 0`)
- Do partial clone to truncate the amount of info (`git clone --depth 1 <repo_URI>`)

I think there should be a "cleaning" for the stale branches? The build for earlier releases is still on the registry so I think the stale branches could be closed.

**Version**
_What version was the issue encountered with_ the repository while cloning, the repository need diet

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
it's the repository, need diet

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
It's the repository, need diet

**Logs**
_Please provide a link to any relevant logs that show the issue_
```
Cloning into 'ardupilot'...
remote: Enumerating objects: 427655, done.
remote: Counting objects: 100% (144/144), done.
remote: Compressing objects: 100% (74/74), done.
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 2544 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
```
There is 427655 objects if we clone directly, when truncated for the main branch only there is 7747 objects! Even after truncating the objects I still get the error, which is my internet fault.
```
Cloning into 'ardupilot'...
remote: Enumerating objects: 7747, done.
remote: Counting objects: 100% (7747/7747), done.
remote: Compressing objects: 100% (6298/6298), done.
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 4729 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
```

sanggusti, 2025-03-28:
Tbh, when I get a stable internet connection it's resolved after 3 tries (the stable internet that I've get is ~3.7 MBps). But maybe this to check the stale branches (when I checked it, most of stale branches last updated is >5 years ago) since I don't think there is use for that.

rmackay9, 2025-03-29:
Thanks for the report.  As you've discovered it's most likely just the internet connection you're using.

In terms of "stale" branches, we have just a few that shouldn't be there, probably mistakes by developers with push access:

- https://github.com/ArduPilot/ardupilot/tree/pr-network-bl-status @peterbarker 
- https://github.com/ArduPilot/ardupilot/tree/revert-25273-matek765se-bdshot @Hwurzburg 
- https://github.com/ArduPilot/ardupilot/tree/Hwurzburg-patch-1 @Hwurzburg 
- https://github.com/ArduPilot/ardupilot/tree/Rover-4.4-Aug31-tests @Williangalvani 

I'll delete these after checking with the above devs.

Txs again, I'll close this now

closed 2025-03-29 by rmackay9