ArduPilot/ardupilot #29660: type infererence to int for AP_xxx types error

tridge, 2025-04-01:
example PR:
https://github.com/ArduPilot/ardupilot/pull/29624

likely also affected:
 - get_minimum_board_voltage() and related fns or maybe get() makes it ok?

tpwrules, 2025-04-01:
Minimum reproducible example:
```c++
#include <cstdio>

template<typename T> class fakeval 
{
public:
    void set(const T &v) {
        _value = v;
    }
    operator const T &() const {
        return _value;
    }
protected:
    T _value;
};

int main(int argc, char** argv) {
    fakeval<float> ff;
    ff.set(0.9);

    float val = argc > 0 ? ff : 0;
    printf("hi %f\n", val);
}

```

tpwrules, 2025-04-01:
I have workshopped some solutions with some very kind folks but it will be a bit before I can clean up and put together a PR.

tpwrules, 2025-04-07:
This is the relatively complete set of affected float casts. I was able to detect them by triggering compiler errors using an older and much worse solution to the problem.

It's possible to cause the same problem with integers (`int16_t v = true ? int16_param : (int8_t)0;`) but I choose to believe it's much less likely to be a problem. Also it's more annoying to fix.

```diff

---
 ArduPlane/Attitude.cpp                       | 2 +-
 ArduPlane/commands_logic.cpp                 | 2 +-
 ArduPlane/pullup.cpp                         | 2 +-
 ArduPlane/quadplane.cpp                      | 6 +++---
 Rover/mode_smart_rtl.cpp                     | 2 +-
 libraries/AP_Camera/AP_Camera_Backend.h      | 4 ++--
 libraries/AP_ESC_Telem/AP_ESC_Telem_SITL.cpp | 2 +-
 libraries/AP_Quicktune/AP_Quicktune.cpp      | 2 +-
 8 files changed, 11 insertions(+), 11 deletions(-)

diff --git a/ArduPlane/Attitude.cpp b/ArduPlane/Attitude.cpp
index e7f5953bb4..349c3ff706 100644
--- a/ArduPlane/Attitude.cpp
+++ b/ArduPlane/Attitude.cpp
@@ -664,7 +664,7 @@ void Plane::update_load_factor(void)
 #endif
 
     float stall_airspeed_1g = is_positive(aparm.airspeed_stall)
-                                  ? aparm.airspeed_stall
+                                  ? (float)aparm.airspeed_stall
                                   : aparm.airspeed_min;
 
     float max_load_factor =
diff --git a/ArduPlane/commands_logic.cpp b/ArduPlane/commands_logic.cpp
index e49b3bc2cb..8997e85f6f 100644
--- a/ArduPlane/commands_logic.cpp
+++ b/ArduPlane/commands_logic.cpp
@@ -1032,7 +1032,7 @@ void Plane::exit_mission_callback()
 #if HAL_QUADPLANE_ENABLED
 bool Plane::verify_landing_vtol_approach(const AP_Mission::Mission_Command &cmd)
 {
-    const float radius = is_zero(quadplane.fw_land_approach_radius)? aparm.loiter_radius : quadplane.fw_land_approach_radius;
+    const float radius = is_zero(quadplane.fw_land_approach_radius)? aparm.loiter_radius : (float)quadplane.fw_land_approach_radius;
     const int8_t direction = is_negative(radius) ? -1 : 1;
     const float abs_radius = fabsf(radius);
 
diff --git a/ArduPlane/pullup.cpp b/ArduPlane/pullup.cpp
index b8cb1a7727..4bf75e35f9 100644
--- a/ArduPlane/pullup.cpp
+++ b/ArduPlane/pullup.cpp
@@ -143,7 +143,7 @@ bool GliderPullup::verify_pullup(void)
     case Stage::WAIT_LEVEL: {
         // When pitch has raised past lower limit used by speed controller, wait for airspeed to approach
         // target value before handing over control of pitch demand to speed controller
-        bool pitchup_complete = ahrs.pitch_sensor*0.01 > MIN(0, aparm.pitch_limit_min);
+        bool pitchup_complete = ahrs.pitch_sensor*0.01 > MIN(0, (float)aparm.pitch_limit_min);
         const float pitch_lag_time = 1.0f * sqrtf(ahrs.get_EAS2TAS());
         float aspeed;
         const float aspeed_derivative = (ahrs.get_accel().x + GRAVITY_MSS * ahrs.get_DCM_rotation_body_to_ned().c.x) / ahrs.get_EAS2TAS();
diff --git a/ArduPlane/quadplane.cpp b/ArduPlane/quadplane.cpp
index c468c2e183..77e5e76a76 100644
--- a/ArduPlane/quadplane.cpp
+++ b/ArduPlane/quadplane.cpp
@@ -2370,7 +2370,7 @@ void QuadPlane::vtol_position_controller(void)
         }
         
         // speed for crossover to POSITION1 controller
-        const float aspeed_threshold = MAX(plane.aparm.airspeed_min-2, assist.speed);
+        const float aspeed_threshold = MAX(plane.aparm.airspeed_min-2, (float)assist.speed);
 
         // run fixed wing navigation
         plane.nav_controller->update_waypoint(plane.auto_state.crosstrack ? plane.prev_WP_loc : plane.current_loc, loc);
@@ -2980,7 +2980,7 @@ void QuadPlane::assign_tilt_to_fwd_thr(void)
[... 49 lines cut ...]
@@ -62,7 +62,7 @@ void AP_ESC_Telem_SITL::update()
         uint8_t motor = bit-1;
         mask &= ~(1U<<motor);
 
-        const float min_rpm = hal.util->get_soft_armed()? sitl->esc_rpm_armed : 0;
+        const float min_rpm = hal.util->get_soft_armed()? (float)sitl->esc_rpm_armed : 0;
         update_rpm(motor, MAX(min_rpm, sitl->state.rpm[motor]));
 
         // some fake values so that is_telemetry_active() returns true
diff --git a/libraries/AP_Quicktune/AP_Quicktune.cpp b/libraries/AP_Quicktune/AP_Quicktune.cpp
index c8c5e5a5e5..5c6101260f 100644
--- a/libraries/AP_Quicktune/AP_Quicktune.cpp
+++ b/libraries/AP_Quicktune/AP_Quicktune.cpp
@@ -518,7 +518,7 @@ const char* AP_Quicktune::get_param_name(AP_Quicktune::Param param) const
 
 float AP_Quicktune::get_gain_mul() const
 {
-    return expf(logf(2.0)/(UPDATE_RATE_HZ*MAX(1,double_time)));
+    return expf(logf(2.0)/(UPDATE_RATE_HZ*MAX(1, (float)double_time)));
 }
 
 void AP_Quicktune::restore_all_params()
-- 
2.47.0
```

PR #29711 (2025-04-28): AP_Param: fix unexpected truncation of AP_Float parameters in expressions
Fixes #29660 for floats at least. An expression like `float v = true ? float_param : 0;` would cause `float_param` to be truncated to an integer to match the other type. Fix by only letting `AP_Float` be converted to a float. It is still possible to get truncation with integer parameter types, e.g. `int16_t v = true ? int16_param : (int8_t)0;` but I don't think it's a problem at this point.

There is a bunch of subtlety and some caveats, please see the commit contents and messages for details. The issue contains a list o [... cut]
closed 2025-04-28 by peterbarker