ArduPilot/ardupilot #29666: NoopLoop ToFSense CAN only 2 sensors working

neurons9, 2025-04-01:
## Bug report

If I connect more than 2 NoopLoop ToFSense to CAN bus, only the first two IDs communicate. All others are not recognized and lead to error messages. Each sensor has an individual ID

If I connect only 2 to CAN and another 2 to serial ports then they all work.

I will test a Cube Orange+ as soon as I have one

The background is that I want to realise obstacle detection in all directions from 4 NoopLoop ToFSense, plus a fifth that points upwards...

**Version**
ArduRover 4.6.0 beta 4
ArduRover 4.5.7 official

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X] Rover
[  ] Submarine

**Airframe type**
Omniwheel and Stearing wheel

**Hardware type**
Cube Black

rishabsingh3003, 2025-04-01:
Hi, can you confirm with NoopLoop what sort of CAN termination resistance each sensor has? 
I wrote the initial driver and only tested with two sensors at most. However, there should be no reason that more don't work. 
The problem you are describing sounds like the CAN Bus may be overterminated.

Ashishbsharma, 2025-04-01:
That's a valid point the issue might be over termination. Each ToFSense sensor could have a built-in 120-ohm resistor, so connecting more than two might drop the total resistance too low. I'd suggest checking if termination can be disabled on the extra sensors and measuring the resistance between CAN_H and CAN_L ideally, it should be around 60-ohm (2 parallel). This might explain why only two sensors work on CAN but all function when some are moved to serial.

peterbarker, 2025-05-01:
Ping @neurons9 we've seen over-termination cause problems in the past.

You can check the static resistance as @Ashishbsharma suggests.

Have you investigated further?

neurons9, 2025-05-06:
That sounds plausible to me. I will carry out further tests and write to the manufacturer. Maybe the configuration software can be extended accordingly.