ArduPilot/ardupilot #29679: AP_Logger: Add MAVLink SYSTEM_TIME like information to Data Flash Log

horiuchi0photo, 2025-04-03:
## Feature request

Currently there is no information about the UNIX time the vehicle is flying, so the GCS or other post-processing tools must infer it from the GPS messages in the log.

I want messages like SYSTEM_TIME to be written to the log when logging is started and when the time base is changed.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2025-05-01:
I've created https://github.com/ArduPilot/ardupilot/pull/29948 which would close this issue AFAICS.

Are you able to test, @horiuchi0photo ?

Note this may end up in a `SYS` log message or similar as we don't have huge numbers of IDs now-adays.

horiuchi0photo, 2025-05-02:
Thank you, this is what I wanted.

I tried the patch, when the time was set during log output, the output was Epoch=0 (it seems the output timing was a bit early).

I think it's fine to add it to another log message.