ArduPilot/ardupilot #29683: Quadplane land detection takes long time

moreba1, 2025-04-03:
**Issue details**
"QuadPlane will recognize that it has landed if the motors are at minimum for 5 seconds and its altitude has not changed more than 0.2m for 4 seconds."

5 second is high for a VTOL drone that have a minimum thrust , also it check 0 value for minimum throttle reach not q_m_spin_min value.

**Version**
4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2025-04-03:
It is this way so we don't disarm while still flying. Quadplanes can hover at zero vtol throttle if there is some wind.

peterbarker, 2025-04-24:
Ping @moreba1 do you understand @IamPete1 's response?

The current behaviour is desired for most users.

Our operations usually involve the GCS operator disarming the vehicle as appropriate (or the pilot disarming-on-a-switch).

peterbarker, 2025-05-09:
Closing this one because things seem to be working as expected.  Please re-open if you think there's something we might need to do.

closed 2025-05-09 by peterbarker