ArduPilot/ardupilot #29723: CAN FD Not working at 5M

ketanatzebu, 2025-04-08:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report
CAN FD Not working at 5M

**Issue details**

_Please describe the problem_
I am trying to get data from a CAN FD sensor, The data rate is stable at 1M CAN FD Config but when I try to run it at 5M, I do not get any data at all.

I am using TI MMWAVE AWR1843 

**Version**
COPTER 4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QuadRotor

**Hardware type**
CubeOrangePlus. CAN1 Port

peterbarker, 2025-05-01:
@ketanatzebu did you enable CANFD on the Cube?

peterbarker, 2025-10-19:
No answer; re-open if that enable didn't work.

closed 2025-10-19 by peterbarker