ArduPilot/ardupilot #29738: AP_Periph+CubeNode DroneCAN Multiframe Message Drops

erbornemeier, 2025-04-09:
## Bug report

**Issue details**

I am running AP_Periph on the head of master on a CubeNode, which is setup to communicate GPS Fix messages over DroneCAN at 1Mb/s on CAN1. Additionally, I have SLCAN enabled on the OTG interface allowing me to connect via the DroneCAN GUI over serial.

The issue I am seeing is that, while I see all messages intending to be sent properly on the SLCAN interface, messages are not actually appearing on the actual CAN bus in full when viewed from a separate socketcan interface tied into the bus directly.

After some detailed investigation, it appears that the STM32H757xx has a configured hw tx buffer size of 7 frames, and the CubeNode is only able to send 7 frames of data out at once when several messages (of the same or different multiframe type) are queued in quick succession.

Here is an image of the gnss messages that are intending to be sent at 5Hz on the bus. The left side is the SocketCAN bus monitor, and the right side is the SLCAN bus monitor on the CubeNode. This is not isolated to gnss messages either, I see similar behavior on the GetNodeInfo message types.

[image]

[image]

Enabling the CAN stats also shows zero cause for concern, and after reading the code, it seems like drops this severe should be incrementing either the tx errors field. There are also no error frames on the bus when viewed over socketcan. A different SLCAN interface on a CubeOrange on the same bus shows exactly what the socketcan interface does as well. The bus itself also looks healthy on an oscillocope and is properly terminated / topology is in order.

[image]

I am suspecting at this point that there is a bug within the AP Periph firmware as I am struggling to find any reason otherwise, but given where the implementation is at today it appears like this has been shown to work by others.

Any ideas / help / confirmation on any of the above would be much apperciated as I feel I’ve exhausted my custom firmware tweaks at this point to try and get this working.

**Version**
Head of master plus some additions to hwdef.dat for CubeNode to enable GPS, moving baseline, SLCAN, serial remaps.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
[X] AP_Periph

**Hardware type**
CubeNode soldered to simple breakout board

erbornemeier, 2025-04-11:
A conversation around this is happening in the ap_periph discord channel - but as an update to connect things here:

I've seemingly found a workaround today. If I alter the `processTx()` function to only send one message (`break;` on successful send), and then alter the `HAL_PERIPH_LOOP_DELAY_US` to `1024` (default is `64` for the STM32H7) in the `hwdef.dat`, I'm seeing all the messages come out on the bus. I suspect this is less of a fix and more of a workaround since I'm effectively just throttling the sends to every 1ms or so which presumably gives the transceiver time to get messages out of the hw queue. Hoping that is good enough for now while hopefully a better fix is found.

Ashishbsharma, 2025-04-19:
A potential enhancement would be to add a check in processTx() function to see if the TX FIFO is full before sending the next frame, perhaps using HAL_CAN_TxReady() or something similar. This could help prevent dropped multi-frame messages instead of relying only on loop delay throttling.  Happy to test if needed.