ArduPilot/ardupilot #29747: WDG Error   Internal Error  0x800

JetPilott, 2025-04-10:
WDG:  T41 SLO FL217 FT3 FA2    Internal Error 0x800

I was flying with a Matek F765 Wing flight controller that has been in the same plane and flown a LOT with no issues for several years.  45 minutes into this flight, the flight controller reset, and posted the following things on the screen before locking up and filling the entire screen with strange characters.  See pictures 

[IMG] https://i.imgur.com/S9XpjrP.jpg [/IMG]
[IMG]https://i.imgur.com/nPDXZWj.jpg[/IMG]
[IMG]https://i.imgur.com/Mq7saVS.jpg[/IMG]
[IMG]https://i.imgur.com/tfiipGz.jpg[/IMG]

Ardupilot 3.5.7 Official Stable

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_ Glider 

**Hardware type**
_What autopilot hardware was used? (Matek F765 Wing)_

**Logs**
_Please provide a link to any relevant logs that show the issue_:   See Pictures

Ryanf55, 2025-04-19:
Sorry, ArduPilot 3.5.7 is no longer supported. Without logs, I'm afraid there's not much we can do.

Ryanf55, 2025-04-19:
Duplicates https://github.com/ArduPilot/ardupilot/issues/29748

closed 2025-04-19 by Ryanf55