ArduPilot/ardupilot #29748: Internal Error 0x800    + WDG Errors

JetPilott, 2025-04-10:
WDG: T41 SLO FL217 FT3 FA2 FORCING LOGGING FOR WATCHD Internal Error 0x800

I was flying with a Matek F765 Wing flight controller that has been in the same plane and flown a LOT with no issues for several years. 45 minutes into this flight, the flight controller reset, and posted the following things on the screen before locking up and filling the entire screen with strange characters. See pictures

[IMG] https://imgur.com/S9XpjrP [/IMG]
[IMG][https://i.imgur.com/nPDXZWj.jpg[/IMG]
[IMG][https://i.imgur.com/Mq7saVS.jpg[/IMG]
[IMG][https://i.imgur.com/tfiipGz.jpg[/IMG]

Ardupilot 3.5.7 Official Stable

Platform
[ ] All
[ ] AntennaTracker
[ ] Copter
[X ] Plane
[ ] Rover
[ ] Submarine

Airframe type
What type of airframe (flying wing, glider, hex, Y6, octa etc) Glider

Hardware type
What autopilot hardware was used? (Matek F765 Wing)  NO SD Card inserted

Logs
Please provide a link to any relevant logs that show the issue: See Pictures

rmackay9, 2025-04-12:
Hi @JetPilott,

Thanks for the report.  Normally we ask that issues are posted first in the forums ([Plane-4.5 forum is here](https://discuss.ardupilot.org/c/arduplane/plane-45/176)) but in any case, could you provide an onboard log?  Also there should be a crash dump file on the SD Card as mentioned on [this wiki page](https://ardupilot.org/copter/docs/common-watchdog.html).

peterbarker, 2025-04-24:
@JetPilott could you provide a copy of your parameters, please?

peterbarker, 2025-05-01:
Ping @JetPilott