ArduPilot/ardupilot #29759: IOMCU - surfaces not moving

tridge, 2025-04-11:
We have had a report of control surfaces not moving on IOMCU connected servos in a pre-flight check.
@peterbarker and I have also seen this once on an aircraft in Canberra, although we did not capture a log (we had assumed it was a hardware issue and were very surprised to find a reboot fixed it).
My current guess is that one of the configuration registers in the IOMCU is not being transferred correctly. For example, if PAGE_REG_SETUP_CHANNEL_MASK was transferred incorrectly then it would cause exactly these symptoms, as it would disable some channels.
In the log from the affected user SERVO_GPIO_MASK is set to 15360, which is Servo 11|Servo 12|Servo 13|Servo 14. A single bit corruption in this value would cause the symptoms of one aileron not moving.
I think we need some way of confirming the correct configuration:
Maybe a 32 bit CRC of all expected config registers and re-send if not correct? Maybe only do this when disarmed?

rmackay9, 2025-04-12:
I've added this to the 4.6.0 issues list

tpwrules, 2025-04-12:
I understand an 8 bit CRC will detect any single bit error. It seems like an implausible cause for this problem.

tridge, 2025-04-15:
> I understand an 8 bit CRC will detect any single bit error. It seems like an implausible cause for this problem.

it would have to be a 2 bit error. With line noise this is unlikely I think, but what about clocking error?

tpwrules, 2025-04-16:
A clocking error should cause UART framing errors too. Do we have any way of recording errors on the IOMCU link so that we can see if affected boards are seeing caught errors?

I recognize this particular message is sent once, but lots of messages are sent to fly the aircraft and uncaught corruption in any of them could be bad. But they should also occasionally get errors too if there is a problem with the link.