ArduPilot/ardupilot #29786: Rover: Follow mode odd circling

stephendade, 2025-04-15:
## Bug report

When in Follow mode, if the leader vehicle is stationary for some time (20-120sec ... it can vary) the following vehicles will begin circling their offset point:

[image]

Can be replicated in SITL with ``sim_vehicle.py --vehicle=Rover --console --map --count=4 --mavproxy-args='--cmd="set fwdpos True"' --auto-sysid``

Then setting for vehicles 2-4:
```
FOLL_ENABLE=1
FOLL_SYSID=1
FOLL_OFS_X=<any>
FOLL_OFS_Y=<any>
```

**Version**
Current master, SITL

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  X] Rover
[  ] Submarine

peterbarker, 2025-05-01:
This feels like the Rovers trying to get somewhere but can't turn that tightly?

rmackay9, 2025-05-01:
I imagine that it's I-term build-up so it might be resolved by perhaps not trying to move if the target is less than the turning radius from the vehicle.

timtuxworth, 2025-05-01:
If it's working the way I think it is, the code is just dropping a WP for the follow vehicle to fly to, it keeps dropping new "pins" head as the target moves, so it never catches up, but if the target stops moving the pin will stop moving and standard "loiter at WP" logic kicks in, which is a circle around the point.

On Plane, I found this behavior so annoying that I stopped using it altogether and I'm now directing the follow vehicle using heading, not WP.

stephendade, 2025-05-02:
Thanks for the ideas!

This had slipped off my radar, but I should be able to investigate further next week.