ArduPilot/ardupilot #29819: Plane: TILT angle only logged for FL and FR

LzREngineering, 2025-04-19:
## Feature request

Request based on 4.6.0 beta 5.

In the logs of a Tilt Rotor VTOL plane, three values are currently recorded for TILT servos: .Tilt (overall) .FR and .FL

Whilst this is sufficient for tri-tilt rotors, in cases where there are more than two tilt servos there may be some benefit in recording the demanded tilt angle from every tilt servo (as defined by Q_TILT_MASK?). Would it be possible to have angular values recorded for all tilt servos eg TILT.RL and TILT.RR (possibly more?)

I know it's also possible to derive the angle from the RCOU log message using the servo max/min pwm parameters - as an alternative, could a new preset be added to UAV Log viewer to enable this?

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ /] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2025-04-19:
The front and rear are the same in all VTOL modes, the only difference is if you have forward flight vectoring enabled.

LzREngineering, 2025-04-19:
> The front and rear are the same in all VTOL modes, the only difference is if you have forward flight vectoring enabled.

I thought that may be the case. So to confirm, TILT.FL=TILT.RL and TILT.FR=TILT.RR? Or to put it another way, all tilting motors on one side of the plane will tilt by the same amount. Under what circumstances would forward flight vectoring created a difference front to rear? looking at my logs, all servos seem to be tilting forwards by 'roughly' the same amount during forward flight. If it can be confirmed that this is alway the case, then I agree this request is not necessary. 
However, could the description for these log meassages in the documentation be updated to reflect the fact that it's the angle of all left side/all right side tilt motors?

IamPete1, 2025-04-19:
> Under what circumstances would forward flight vectoring created a difference front to rear?

`Q_TILT_FIX_GAIN` none zero. Then the front and rear tilts will differentially vector in forward flight for pitch control.

LzREngineering, 2025-04-19:
> > Under what circumstances would forward flight vectoring created a difference front to rear?
> 
> `Q_TILT_FIX_GAIN` none zero. Then the front and rear tilts will differentially vector in forward flight for pitch control.

In that situation, would the rear tilts remain locked at zero (fully forward)? - if not, there may still be a benefit in recording TILT.RL and TILT.RR meassages.

IamPete1, 2025-04-19:
> In that situation, would the rear tilts remain locked at zero (fully forward)? - if not, there may still be a benefit in recording TILT.RL and TILT.RR meassages.

No, to pitch up for example the fronts would tilt up and the rears tilt down.

peterbarker, 2025-05-01:
Ping @LzREngineering - has Pete convinced you our logging is sufficient?

LzREngineering, 2025-05-31:
> Ping [@LzREngineering](https://github.com/LzREngineering) - has Pete convinced you our logging is sufficient?

IMO as there are situations where tilt can vary front-to-rear as well as left-to-right I still think it would be useful to have TILT logged for each tilting motor. However I woudn't see it as a high priority as this information can be derived in post-analysis.

One thing I was looking to do was display the TILT on the OSD - not sure if that is currently possible?

IamPete1, 2025-06-01:
I think I will close this, as you say it can be worked out in post analysis. Forward flight vectoring is not widely used.

No, there is no way to display tilt on the OSD.

closed 2025-06-01 by IamPete1