ArduPilot/ardupilot #29820: Warn if using BLHeli 32.9 or lower

Ryanf55, 2025-04-19:
## Feature request

**Is your feature request related to a problem? Please describe.**

BLHeli 32.10 has fixed a major issue with unintentional motor spoolups.

I've seen multiple instances of spoolups, or disarming and the motors not stopping becuase they detect the wrong protocol.

This is a major safety issue for anyone using BLHeli 32.9 or less.

**Describe the solution you'd like**

On bootup, ArduPilot can try to communicate with DSHOT to retrieve the blheli version. If the version is not 32.10, a warning can be issued. ArduPilot could the version and warn users that if they fly 32.9 or less, they need to throw their ESC's away and replace them with something safer.

**Describe alternatives you've considered**

Not warning users their ESC's are dangerous.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

https://github.com/bitdump/BLHeli/issues/743

https://github.com/bitdump/BLHeli/releases/tag/Rev32.10

[image]

hendjoshsr71, 2025-04-20:
> bitdump/BLHeli#743 <https://github.com/bitdump/BLHeli/issues/743>

Is there anyway to update BLHeli32 now that the servers are down? I thought
not…..

IE did someone find a workaround?

Ryanf55, 2025-04-20:
Nope. It's my understanding  that it's unrecoverable. I saw a translated-chinese-forum that some people are trying to crack the server keys, but they haven't achieved it yet.

andyp1per, 2025-04-20:
All I can say is that I have never encountered this problem myself, despite having multiple 32.9 ESC's. I am not sure what the difference is.

Ryanf55, 2025-04-20:
Until I can come up with a reproducible way to cause a spool up, I'm considering this on hold as far as a code implementation.
We could at least update the wiki.

peterbarker, 2025-05-01:
Ping @Hwurzburg on the Wiki concept.

Hwurzburg, 2025-05-01:
Added wiki warning