ArduPilot/ardupilot #29836: AP_DDS: Support throttle surpress in GUIDED mode for plane

Ryanf55, 2025-04-20:
## Feature request

**Is your feature request related to a problem? Please describe.**

I can't use GUIDED mode to turn off the throttle on a motor glider using either mavlink or DDS.

**Describe the solution you'd like**

Guided should respect overrides so you can surpress throttle.

**Describe alternatives you've considered**

Adding offboard mode

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

I want to ensure the motor is off, commanding target speed does not do that.

peterbarker, 2025-05-01:
The referenced issue https://github.com/ArduPilot/ardupilot/issues/4337 doesn't have anything to do with servo outputs.  Overriding your throttle to be zero won't make an inch of difference.

I'm not sure this is something that should go into "guided".  Perhaps into that new "offboard" mode I've been hearing about?

Ryanf55, 2025-05-09:
Approach discussed today: set_gliding_state ROS -> AP_DDS-> AP_ExternalControl -> AP_ExternalControl_Plane -> plane.TECS_controller.set_gliding_requested_flag