ArduPilot/ardupilot #29845: Vehicle incorrectly reports waypoint reached after flying continuously away from it

wyunc, 2025-04-21:
## Bug report

**Issue details**

I loaded a square mission identical to issue  #26519. When I set the parameter MOT_PWM_MIN to 2000, the copter continuously moves forward, getting progressively farther away from the waypoint, which is expected behavior. However, after traveling some distance in this direction, ArduPilot suddenly and incorrectly declares that the waypoint has been reached and switches to the next one. After continuing for some time, it again mistakenly marks this next waypoint as reached.

I suspect this issue is similar to the "fast waypoint" bug described in issue #26519. I've confirmed that the previously buggy code from #26520 has been patched in the version of ArduCopter I'm currently testing, yet the problem persists.

Additionally, setting the parameter MOT_SPIN_MIN to 1 also triggers the same bug.

Perhaps there is some undiscovered default value accumulating when calculating the distance from the vehicle to the waypoint reference point?

**Version**
4.7.0-dev

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2025-04-21:
Hi @wyunc,

Thanks for the report.  It's funny that we merged PR https://github.com/ArduPilot/ardupilot/pull/26520 but didn't close issue https://github.com/ArduPilot/ardupilot/issues/26519.  I suspect that was just an administrative oversight.

This report does definitely seem like a duplicate of https://github.com/ArduPilot/ardupilot/issues/26519 but your testing indicates that we may not have fixed the bug completely

wyunc, 2025-04-21:
Hi @rmackay9 , the same issue also occurs in Rover. Setting the parameter ATC_SPEED_FF to values such as 10000.0, 1200.14, or 87.07 can trigger this problem. The detailed behavior is as follows:

I set a navigation mission to travel towards WP2 and adjusted the ATC_SPEED_FF parameter accordingly. On its first pass, the rover physically reached WP2 but didn't recognize it had done so. Instead, the rover continued moving forward and began circling at a location far away from the waypoint. Similarly, after a certain amount of time, ArduPilot incorrectly reported WP2 as reached, then switched navigation to the next waypoint (WP3). However, the rover continued circling at the same distant location and, again after some time, incorrectly reported WP3 as reached—even though it never approached WP3.

[image]

[image]

rmackay9, 2025-04-22:
Hi @wyunc,

Sure, feel free to send me the google document.  I'm also on discord as "rmackay9".  One thing though, of course it's very possible to cause a vehicle to crash or fly very badly by setting various parameters to incorrect, invalid or out-of-range values.  We don't guarantee that a vehicle will operate correctly even if parameters are set incorrectly.

peterbarker, 2025-04-29:
Ping @wyunc - request for access to that document from Randy

rmackay9, 2025-04-29:
Hi @peterbarker,

BTW I've seen the document now and given my feedback to wyunc.  The short-form of the feedback is that there are probably some real issues found but many are issues that would only appear in SITL or are simply parameter configurations that don't match the vehicle (e.g. setting the motor ranges incorrectly)

wyunc, 2025-04-29:
HI @peterbarker ,
Yes, I discussed this with @rmackay9  , and he offered some valuable insights. To summarize, some inputs do indeed affect the vehicle's outcomes, but this falls outside the scope of AP's considerations. That said, I genuinely welcome and appreciate diverse perspectives and suggestions.
Could @rmackay9  to share this link with @peterbarker ? I’m not sure how to send it privately otherwise. Thanks!

peterbarker, 2025-05-01:
So I've been trying to reproduce this without luck.

@wyunc when you say, "getting further from the waypoint", is this moving directly away from the waypoint?

ArduCopter can consider a waypoint complete *even if you are getting further away from the waypoint* if you cross the "finish line".  Here's an example of a mission which progresses once the vehicle passes the finish line, even 'though the vehicle is getting further from the current waypoint: 

[image]

(this is not a bug, it's a feature ;-) )

When travelling 3->4 there's a finish line perpendicular to the edge between 3 and 4 (in this case)

I've created https://github.com/ArduPilot/ardupilot/pull/29945 which you can use to reproduce the above image.

I'm unable to reproduce your problem given your instructions, *apart* from this finish-line mechanism.

@wyunc could you provide more details about this "completes waypoint even 'though it's getting fruther away" problem, please?  It would be nice if you could modify the autotest I've created in that PR to reproduce the problem you're seeing - so when we fix it it stays fixed!

wyunc, 2025-05-01:
Hi @peterbarker ,
I encountered this issue during a northbound flight after takeoff from “home” when I set MOT_PWM_MIN=2000 or MOT_SPIN_MIN=1: the copter kept flying north and moving away from its waypoints. You can see exactly what I mean in the image in issue #26519. Here’s what happens:
1.After updating the parameter, the copter flies past WP2 to the north but does not register it as reached.
2.It continues drifting further north and, after some time, suddenly reports that WP2 has been reached.
3.It then points to WP3, and again, after a while, reports WP3 as reached before switching to WP4.

This behavior perfectly matches #26519—waypoints are marked complete even though the copter never actually reaches them.

I’ve attached the mission file I used for this test.

[case1.txt](https://github.com/user-attachments/files/19998630/case1.txt)

I’m sorry for any inconvenience—I’m currently on an extended holiday without access to a computer running AP. If you don’t mind waiting, I’ll be back at work on May 7th and can send you detailed screenshots then.

wyunc, 2025-05-07:
Hi @peterbarker ,
[Here](https://drive.google.com/file/d/1RGtqSY1JrCBlWfb5cpLJgUE0IMihVSlf/view?usp=sharing) is the video I recorded—feel free to jump straight to the one-minute mark to start watching. After takeoff, I set **MOT_PWM_MIN=2000**, and you’ll see the copter fly past the target waypoint without registering it as reached. However, after a delay time **T**, it suddenly reports the waypoint as reached—and that delay **T** is different for each waypoint.

rmackay9, 2025-05-07:
Setting MOT_PWM_MIN = 2000 results in the motors going to 100% at all times which probably results in the vehicle climbing very very high.  I suspect the vehicle being impossibly high is why we have trouble recreating the issue.

lthall, 2025-05-08:
Could you please provide a log?

wyunc, 2025-05-08:
Hi @lthall, Sure! [Here's ](https://drive.google.com/file/d/1NqblxBKdPUjIKCVG12mSf9yBMOP2gIcY/view?usp=sharing)the log file. Please feel free to let me know if you need anything else!

lthall, 2025-05-09:
Ok, the first thing is we are looking at an aircraft that is unable to control it's position, so the only thing we are looking at here is that the fact that fast waypoints don't currenlty check the instantainous distance to the waypoint to declare we have reached the waypoint.

Fast waypoints are treated as just one point on a kinimatic path. As such the WP_Nav libary manages the position along the kinimatic path, through fast waypoints, to ensure the target position doesn't move further along the path than the aircraft is currently positioned. This is done correctly here.

This is confusing because while the aircraft position is not under control, it is drifting in the general direction of the specified path. As such the instantainouos target along that path is moved only slightly in front of the minimum cross track distance. 

This will continue until the direction of the path, and therefore the target location, moves away from the current aircraft location. At this point the speed along the track will decay exponentially towards zero. As this is an exponential decay it won't ever reach zero so we will still move towards the next waypoint, but at an ever slower rate. This is the change made in #26520, previously it could only slow down to 10%.

I wouldn't call this a bug as the code is working exactly as we intend. Instead I would ask some questions:
- Do we want to slow down the progression along the path if the cross track error becomes large compared to the waypoint radius. 
- Should we change the slowing along the path to a liniar rather than exponential decay so we do come to a complete stop.
- Should we abort a mission and call some sort of failsafe if we can't control position past some threashold. (especially as this is required to demonstrate this behaviour)

I would suggest we close this issue as it is more realisticly shown in #26519.

rmackay9, 2025-05-09:
OK, I'll go ahead and close this issue.

Thanks @wyunc for the report and this does confirm that we still have an issue.

closed 2025-05-09 by rmackay9