ArduPilot/ardupilot #29854: Pixhawk 6x rev8, holybro micro gps with IST8310 external compass not detecterd

drbios, 2025-04-21:
## Bug report

**Issue details**

Holybro Micro GPS M9N and M10 with IST8310, the  external compass is  not detected, tryed on GPS1 nad GPS2 port, only internal compass is detected
GPS work perfect but compass not detected

**Version**
Ardupilot 4.5.7, 4.6 and 4.7 
pixhawk 6x  on mini base
Pixhawk 6x (rev.8)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
queadcopter X frame

**Hardware type**
holybro pixhawk 6x rev 8, pixhawk  mini base, holybro micro gps M10 and Holybro M9N

**Logs**

HB-SP-P, 2025-04-22:
Do you have another GPS unit or flight controller/baseboard to test the I2C compass? I could not replicate this issue on my end.
You can also contact Holybro support for any potential hardware issues.

drbios, 2025-04-22:
> Do you have another GPS unit or flight controller/baseboard to test the I2C compass? I could not replicate this issue on my end. You can also contact Holybro support for any potential hardware issues.

Hello, I have 2 Micro GPS units, a Micro M10 and a Micro M9N, both have the same compass and the compass is not detected on either of the 2 GPS units.

peterbarker, 2025-04-29:
> Hello, I have 2 Micro GPS units, a Micro M10 and a Micro M9N, both have the same compass and the compass is not detected on either of the 2 GPS units.

The suggestion from @HB-SP-P was to try the units on a different flight controller :-)  Do you have another flight controller you can try the two GPS units on?

Additionally, do you have a known-working GPS/i2c combo you can test on the Pixhawk6X, or are those two the only 2 available to you?

peterbarker, 2025-05-01:
I've marked this "Bug" because it definitely should be found.

At the moment I'm guessing  the bug's actually on @drbios  end :-)

rmackay9, 2025-05-05:
@drbios,

Any feedback?  Testing with another autopilot will prove whether it is a hardware issue with the one autopilot or not