ArduPilot/ardupilot #29880: Firmware Build Server: rename QURT .deb package to exclude githash

rmackay9, 2025-04-24:
I'm working on updating our VOXL2 installation instructions and I notice that the .deb package we produce includes the githash in the filename.  E.g. voxl-ardupilot_ArduCopter_4.7.0-39ac45a7_arm64.deb.

See latest here: https://firmware.ardupilot.org/Copter/latest/QURT/
See beta here: https://firmware.ardupilot.org/Copter/beta/QURT/

We don't do this for any other binaries that we build and it makes it slightly more difficult for us to produce copy-paste-able commands that the user can use to implement the installation

I'm sure we can craft copy-paste-able commands but I wonder how people feel about simply removing the githash.  Surely there are other ways to know if the binary is up-to-date or not

hendjoshsr71, 2025-04-25:
I prefer githash info.  When I build and send firmware to folks I always
include it. Yes, there are other ways to get it but it is easier for
tracking and for the end user.

  If you want easy commands to copy/paste for the unique bits, replace
those with a wildcard character???

Honestly, it is super annoying that our tooling spits out generic
ArduCopter.apj files when you build for STM32. I must rename those before
sending them to a user with board type and version at a minimum. I've
been meaning to create a PR that even adds in branch name to the APJ file
for logging purposes.

As a dev when you've got 5 versions of the app image you are testing
between having that extra info in the build makes a large difference.
Especially, on VOXL2 where your debugging capabilities is in the stone ages
of log files and printf statements.

timtuxworth, 2025-04-25:
I agree with Josh - this naming convention with the vehicle type, branch and hash is awesome and IMO we should be doing it for all builds, not removing it from here.

peterbarker, 2025-04-29:
@rmackay9 would a `-latest` link work for you?

eg.

`voxl-ardupilot_ArduCopter_4.7.0-latest_arm64.deb` -> `voxl-ardupilot_ArduCopter_4.7.0-39ac45a7_arm64.deb`

assuming that 39ac45a7 was the latest build....

rmackay9, 2025-04-29:
Hi @peterbarker,

Yes, that would be fine with me

I'm happy to go along with what the majority of people want but I think we should be consistent.  It just seems odd to me that this is the only build like this.

peterbarker, 2025-04-29:
```
Package: voxl-ardupilot
Version: FW_VERSION
Section: devel
Priority: optional
Architecture: arm64
Depends: modalai-slpi(>=1.1.19), libslpi-link
Maintainer: Eric Katzfey <eric.katzfey@modalai.com>
Description: ModalAI ArduPilot flight controller
```

Looks like the metadata in the deb isn't right, @katzfey .  That `FW_VERSION` is odd.

rmackay9, 2025-04-29:
Due to objections I'm happy to leave the name as-is

closed 2025-04-29 by peterbarker

katzfey, 2025-04-29:
@peterbarker That FW_VERSION is a placeholder that gets overwritten with the actual version string in the libraries/AP_HAL_QURT/packaging/make_package.sh script when creating the package.