ArduPilot/ardupilot #29881: Extra parameter - SMBUS Generic

adhit-n, 2025-04-24:
Request
1. Display cell voltages of the second SMBus Generic monitor after all the available cell voltages has been displayed.
What I mean here is that, for example, say we have two SMBus Generic battery monitors where each monitors 6 cell voltages. We have noticed that only one of the battery pack's cell voltages are being displayed though there are two monitors connected.
2. Display cycle count, consumed mAh and other critical parameters as well.

We have noticed that the Ardupilot firmware reads many parameters but those are not defined in Mission Planner GCS.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

VermennikovMV, 2025-04-27:
ArduPilot (4.3+) publishes two BATTERY_STATUS messages over MAVLink - id = 0 and id = 1 - each containing an array of 6 (or up to 12) cell voltages.
The second battery data is ‘lost’ to Mission Planner if at least one of the conditions is met:
   1. Identical BATT_SERIAL_NUM (i.e. equal to the same number): the firmware decides that the monitors belong to the same battery 
   and writes both cell portions to the same pack.
   2. Old MP build (< 1.3.82): before patch [#3105](https://github.com/ArduPilot/MissionPlanner/blob/master/ChangeLog.txt?utm_source=chatgpt.com) the HUD interface had only one Battery widget and ignored the id field, so the 
   second batch would overwrite the first (there was just a bug that Batt1 was sending packets with the same id as Batt2).
   3. New MP, but the widget is not enabled: after the edit, the ‘Battery 2’ widget needs to be ticked on the HUD (see screenshot).
[image]

For example, these are the settings I have set:
BATT_MONITOR = 7
BATT2_MONITOR = 7
BATT_SERIAL_NUM = 1
BATT2_SERIAL_NUM = 2 

Additionally MP does not have fields for cycle count yet, so even the latest version does not draw them, although the firmware passes them. If you want such functionality to be added, I can make a commit to the MissionPlanner repository with a proposal for new functionality.

adhit-n, 2025-04-28:
Hi VermennikovMV
I think I have not conveyed the issue in the right manner.
Please see the screenshot.
[image]

I have connected two 6s Battery monitors to the FC. And all the parameters are being shown correctly in the HUD as well as in the Status Tab.
The issue that we are facing is, we are able to see cell voltages of the first monitor BAT1 only in the status tab. The cell voltages of BAT2 are not being shown anywhere. What we are expecting is, the cell voltages from the second monitor should start from Cell 7-Cell 12.
Is it possible to initiate this change in MP or should I change some configuration in MP? 

Also it would be great if can make a commit to the MissionPlanner to add Cycle Count field.

peterbarker, 2025-04-29:
> Is it possible to initiate this change in MP or should I change some configuration in MP?

I would imagine so.  All of the data is being provided by the autopilot, so I'm really not sure why you raised this issue against the ArduPilot firmware repository rather than the MissionPlanner repo (github.com/ardupilot/MissionPlanner)

OK to close this one?

adhit-n, 2025-04-29:
Apologies, Peter. I'll close this issue here and raise it in the MissionPlanner Repo.
Thanks for your feedback.

closed 2025-04-29 by adhit-n