ArduPilot/ardupilot #29914: Plane: Guided Fly To not safe in steep terrain and SITL doesn't detect crash

Ryanf55, 2025-04-28:
## Bug report

**Issue details**

Using the MAVProxy "Fly To" command can crash the vehicle in steep terrain because it never ascends to gain altitude.

**Version**
ArduPilot-4.6, hash b07d7f8b28bcd6dd2d23e34e11e59b42d754b20a

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
N/A

**Steps to reproduce

```
./Tools/autotest/sim_vehicle.py -v plane --console --map --custom-location="21.446158, -157.862868, 100, 75"
param set TERRAIN_FOLLOW 1
arm throttle
mode takeoff
```

Once flying, click a FlyTo point SSW the vehicle across the mountain range.

Observe the vehicle accept the command, and fly right into the side of the mountain.
SITL does not seem to detect the crash either, which is a bug.

```
![Image](https://github.com/user-attachments/assets/dab54495-dcdd-4e48-907f-ef6d5fd04c36)
![Image](https://github.com/user-attachments/assets/4dde3160-c270-4777-bc4c-03a86d2d25bc)

**Logs**

[79.zip](https://github.com/user-attachments/files/19941937/79.zip)

timtuxworth, 2025-04-29:
This is a known bug, specifically the Terrain lookahead function doesn't correctly detect high terrain between the current location and next WP. This will also happen in an AUTO mission.

In my "Terrain Avoid" Lua script, I re-implemented the terrain lookahead to detect this and then to react by:
1. Attempting to pitch up over the "hump"
2. If that doesn't work, attempting to "loiter up" to gain altitude to get over the hump. This looks left/right to pick the best way to go to avoid hitting other high terrain.
3. Worst case, engaging the VTOL motors to quickly gain altitude

This only works with a quadplane, but except for #3, this would also apply to a regular fixed wing.

My PR for this is pending real world testing. I've tested it extensively in SITL.
#28625

timtuxworth, 2025-04-29:
In my experience SITL does detect it and reports "hit ground".

timtuxworth, 2025-04-29:
I worked extensively with @Georacer on this https://github.com/ArduPilot/ardupilot/pull/28891 which attempts to give the vehicle a better chance of flying over high terrain. My conclusion is that the current behavior is probably as good as it gets.

Zandriy, 2025-05-21:
> In my experience SITL does detect it and reports "hit ground".

I've also reproduce the issue, it makes "hit ground" if it flies to NW or W, but if it flies to SSW or SW than at some point it stays almost on the same place and increases height, like it is a copter.

[image]