ArduPilot/ardupilot #29934: SITL with QGroundControl | sim_vehicle | Own board - through Serial

gg-bt, 2025-04-30:
Hello everybody.

MY GOAL
I want to run a SITL where I control my simulated copter through mavLink messages sent from a MCU through a serial port.
I was already able to have the capability of controlling it using python thanks to this [nice tutorial](https://github.com/Intelligent-Quads/iq_pymavlink_tutorial/blob/master/movement.py) from Intelligent Quads, I am trying to transpose the same setup to my current scenario.

WHAT I AM DOING
I checked already that my C code running in the MCU can send and receive properly mavLink messages.

Then i run sim_vehicle in this way

sim_vehicle.py -v ArduCopter --out=/dev/ttyUSB0,19200

where the --out is my USB/Serial port running at 19200 bps, my board with the MCU is transmitting and receiving on this port.
Then i run QGroundControl because my ultimate goal is to display in the GUI the actions driven by the commands I am sending from the MCU. If I am not wrong, when i run the sim_vehicle.py by default i will have one --out UDP connection with the QGC. In QGC i can see data coming in which should be the data from the sim_vehicle.py.
At this moment I have three instances running:

-     my ArduCopter in a shell where i can see the commands received and executed by ArduCopter
-     my ecplise with my code running and receiving the heartbeat
-     QGC connected to the sim_vechile.py

QUESTIONS
I want to, as first, set the home position of the simulated vehicle through my MCU. I used the this code after being sure to get the heartbeat from the vehicle

```
uint8_t thisSystemId    = 255;
uint8_t thisComponentId = 1;

mavlink_msg_set_home_position_pack( thisSystemId, thisComponentId, &mavMsg, target.systemId, -36.359979, 149.164565, 60, 0, 0, 0, NULL, 0, 0, 0, 10);
```

I cannot see anything happening in my ArduCopter shell neither in my QGC.
Of course I tried the example mentioned above about Intelligent Quads in this way

```
uint8_t thisSystemId    = 255;
uint8_t thisComponentId = 1;

mavlink_msg_set_position_target_local_ned_pack(thisSystemId, thisComponentId, &mavMsg, 10, target.systemId, target.componentId, MAV_FRAME_LOCAL_NED,0x0DF8,10,0,-10,0,0,0,0,0,0,0,0);
```

after receiving the heartbeat and manually setting the vehicle in guided mode and after a take off as in the example as well, no luck. Target component and system id are both equal to 1.
I would like to understand what I am doing wrong and why:

    I don’t see my arducopter receiving any instruction from the MCU or at list refusing the message sent ( I am referring to the set home)
    What could be a possible test to be done to see if the simulated vehicle is actually receiving ifo from my MCU?
    Is the way i am launching the SITL correct in terms of setting the serial port communication?
    How i would add to the picture the capability to connect to QGC the simulated vechile and see how it reacts to the mavlink commands sent through the MCU

BOTTOM COMMENT
It is a long post but i wanted to add as many details as i could to try to get some help. Feel free to ask more if needed, in the meantime, thanks to you all.

rmackay9, 2025-05-01:
Hi @gg-bt,

So you're trying to connect a real sensor/device to a simulated vehicle.  We have some info here https://ardupilot.org/dev/docs/learning-ardupilot-uarts-and-the-console.html

The simulated vehicle's location can be changed when SITL is started.  Trying using the "-L" command line argument. It can't be changed once SITL has been started.  Moving the home location will change where the vehicle will RTL to, it won't affect the location of the vehicle itself. 

We have this mavlink interface page which includes copy-paste-able commands https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html.  So maybe try using these first?

I'm going to go ahead and close this issue if that's OK because questions should really go in the support forums or on Discord.  Sorry to be strict but if we're not then our to-do becomes another support forum.

Thanks for using AP!

closed 2025-05-01 by rmackay9

gg-bt, 2025-05-01:
It is completely fine to close it, I am struggling a bit with this setup then I wrote this here with the hope of getting some hints or redirection.
I really appreciate, thanks a lot @rmackay9