ArduPilot/ardupilot #29943: Unstable fixed wing to VTOL transition in RTL/QRTL

batmanthemaverick, 2025-05-01:
I am working on a non vectored dual rotor tailsitter which fly well in VTOL and fixed wing also the transition. But whenever i call for RTL/QRTL it has this unstable transition to VTOL. I have checked with FBWA to QLoiter everything works but not in RTL/QRTL. 

Since I am facing this issue from more than 1 year, I assume it to be a bug. 

Link for the forum, https://discuss.ardupilot.org/t/dual-tailsitter/116580/32?u=tailor

**Issue details**
Transition between FW to Qloiter is stable but not in RTL/QRTL

[image]
This if when FBWA to Qloiter

[image]
This image is when RTL

**Version**
I have tried from 4.5.0 to 4.5.7

**Platform**
[  ] Plane

**Airframe type**
Non vectored dual rotor tailsitter

**Hardware type**
_Orange cube plus

**Logs**
https://drive.google.com/drive/folders/1aq9JY4Cd_5jd_P8Lb85R8-0xxI9xO2jk?usp=sharing

@IamPete1 @tridge @magicrub Tagging for your attention

IamPete1, 2025-05-06:
Not a bug, just a limitation of the current implementation. We have a issue for it here: https://github.com/ArduPilot/ardupilot/issues/25771

robustini, 2025-05-06:
It is not a bug but an intended behavior for VTOLs with Ardupilot.
During QRTL as soon as the transition is completed immediately the VTOL starts to drop in altitude while centering on the home.
This is not good for non-vectored tailsitters, it has been widely demonstrated.
There is a fix that I have shown to work well but it needs to be validated and in case improved to the developers in charge of this.

IamPete1, 2026-09-07:
We now have `Q_RTL_PAUSE_TIME` to allow a delay before starting the descent.

closed 2026-09-07 by IamPete1