ArduPilot/ardupilot #29958: Add Zenoh integration

patrickelectric, 2025-05-02:
## Feature request

[Zenoh](https://github.com/eclipse-zenoh/zenoh) is a mature framework (5 years old) maintained by [Eclipse foundation](https://www.eclipse.org/) and [ZettaScale](https://www.zettascale.tech/), it outperforms MQTT, Kafka, and DDS. From a [performance study](https://arxiv.org/pdf/2303.09419), it can achieve “up to 67 Gbps on single-machine tests and 51 Gbps on multiple-machine tests with a 100 GbE network when the peer mode was used. For the largest performance gaps in bitrates, Zenoh even shows one to two orders of magnitude improvement on MQTT and Kafka (more than 50x and 100x in both scenarios) and could double the throughput of Cyclone DDS. For the closest numbers, Zenoh can still be several times the performance compared to the others. For latency, Zenoh has lower numbers than MQTT and Kafka. Although DDS obtains shorter latency than Zenoh does for the single-machine scenario due to its use of the UDP multicast transport mechanism, we can see that the microcontroller implementation of Zenoh, i.e. Zenoh-pico, which has already realized the UDP multicast transport, can perform better than DDS in all cases. This capability will be supported by all Zenoh implementations in the future (2023).”

[ROS2 study for a RWM alternative](https://discourse.ros.org/uploads/short-url/o9ihvSjCwB8LkzRklpKdeesRTDi.pdf), Zenoh was chosen to be the DDS alternative after being compared to [Cyphal](https://opencyphal.org/), DDS, [eCal](https://github.com/eclipse-ecal/ecal), [Kafka](https://kafka.apache.org/intro), [LCM](http://lcm-proj.github.io/lcm/), MQTT, [NNG](https://github.com/nanomsg/nng), OPC-UA, [TCPROS](http://wiki.ros.org/ROS/TCPROS) and [ZeroMQ](https://zguide.zeromq.org/docs/chapter1/).

Zenoh already supports officially [Rust](https://github.com/eclipse-zenoh/zenoh), [C++](https://github.com/eclipse-zenoh/zenoh-cpp), C ([embedded](https://github.com/eclipse-zenoh/zenoh-pico) or [not](https://github.com/eclipse-zenoh/zenoh-c)), [Python](https://github.com/eclipse-zenoh/zenoh-python), REST, [Java](https://github.com/eclipse-zenoh/zenoh-java), [Kotlin](https://github.com/eclipse-zenoh/zenoh-kotlin), [TypeScript](https://github.com/eclipse-zenoh/zenoh-ts) (now stable), [C#](https://github.com/eclipse-zenoh/zenoh-csharp) and [Go](https://github.com/eclipse-zenoh/zenoh-go) (both in hold), with bridges for [ROS](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds)1/2, [DDS](https://github.com/eclipse-zenoh/zenoh-plugin-dds), [MQTT](https://mqtt.org/) and support for multiple storage plugins such as [influxdb](https://www.influxdata.com/), [s3](https://aws.amazon.com/s3), [rocksdb](https://github.com/facebook/rocksdb) and [filesystem](https://github.com/eclipse-zenoh/zenoh-backend-filesystem). It also provides a Wireshark [plugin](https://github.com/eclipse-zenoh/zenoh-dissector) for network development / debug.

**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Zenoh integration in ArduPilot.

**Describe alternatives you've considered**
AP_DDS is a possible alternative.

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

- Just released 1.0, a stable API that will be maintained and supported, also designed with extensions points in the protocol to be backward compatible if necessary.
- Has QoS[1](https://speakerdeck.com/kydos/qos-dds-vs-zenoh),[2](https://docs.rs/zenoh/latest/zenoh/qos/index.html)
- Was projected to be very efficient on the wire, very low overhead, close to 10s bytes, close to bluetooth that uses 8. Topic names are mapped to integers.
Any message is used as a keep alive message, if nothing happens a keep alive is used every 10s.
- There are applications running in Time Sensitive Networking (TSN) [time sliced]

Note: I believe that @khancyr and @Ryanf55 would be interested on that 
Ref: #23424

patrickelectric, 2025-05-02:
Just to add more on that, we did hardware tests using Zenoh on a STM32H5 with great results for an ESC control.

khancyr, 2025-05-02:
Agree.

I got some contact at Zetascale to help if need.

To be noticed that the version that could fit on stm32 is only picozenoh that is a subset of zenoh. That should be compatible with microros agent.
But indeed the capability promise are great and would great improve AP interface to IoT world.

As zenoh is the transport later we could also use mavlink over zenoh and benefits from its advance routing features !

patrickelectric, 2025-05-02:
> As zenoh is the transport later we could also use mavlink over zenoh and benefits from its advance routing features !

That would be great! 
We have a proof of concept integration of zenoh in [mavlink-server](https://github.com/bluerobotics/mavlink-server/tree/master). We are breaking messages and variables to be easily accessible.

[image]

[image]

stephendade, 2025-08-20:
I'd be interested in Zenoh too, as an option in the AP_DDS library - an alternative to MicroXRCE DDS. So it'd be used as an RMW middleware in ROS2 setups.

Reason for this is that Zenoh appears to be far more efficient (compared to other ROS2 DDS offerings) in constrained networks, which ArduPilot vehicle typically run on.