ArduPilot/ardupilot #29975: AP_NavEKF3: Position reset should not use GPS when using ExtNav

rishabsingh3003, 2025-05-05:
Currently, when a position reset is triggered within EKF3 (due to prolonged rejection of the position estimate), the filter resets to a "default" position source:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L835

If GPS is available, it is automatically chosen as the reset source. This behaviour is problematic when flying with External Navigation (e.g., VISO), because the external navigation solution may have significantly drifted from the absolute GPS position. When the reset occurs to the GPS location, any subsequent ExtNav observations are likely to be rejected due to large innovation values.

This results in the EKF continuously resetting its position, preventing proper fusion of external navigation observations and leading to unstable or unusable state estimation.

rmackay9, 2025-05-05:
Thanks very much for this.  This is mentioned in this issue as well https://github.com/ArduPilot/ardupilot/issues/28968

rishabsingh3003, 2025-05-05:
I am hoping the Partner who reported this can give me a REPLAY log so we can fix it, however, this should also be fairly trivial to reproduce in SITL I think

rishabsingh3003, 2025-05-05:
.. Actually this is already solved. I was looking at a slightly older branch. Should rebase before I open issues :)

closed 2025-05-05 by rishabsingh3003