ArduPilot/ardupilot #30058: Flash ECC handling problem on boards where params are stored in flash

rmackay9, 2025-05-13:
During AP-4.6 beta testing we discovered that we have at least one issue with our flash ECC handling on boards where the parameters are stored in flash (e.g. the same place where the firmware is stored).

This issue was discovered on some TBS boards where the user found that occasionally the board's firmware would be cleared forcing the user to reload the firmware.

The bug in our ECC handling is that we check for errors in the parameter area (which as mentioned is also in flash) but then clear the firmware and leave the parameter area untouched.  This means that theoretically the ECC check should trigger again after the firmware is reloaded.  Strangely this is not the case though, the ECC check is not triggering again

Related PR:

- PR fix for 4.6: https://github.com/ArduPilot/ardupilot/pull/30031

ig-raiderler, 2025-05-13:
Interesting solution to this problem! I've been working on similar challenges recently.

rmackay9, 2025-05-13:
I've removed the comments of some AI spammers