ArduPilot/ardupilot #30076: Fixing ExternalNav + GPS

eufrizz, 2025-05-15:
## Bug report
(previously discussed on Discord under the partner-theseus channel)
**Issue details**
Ardupilot is fusing external nav and GPS data when only external nav is selected as the input source.
From my understanding, only one of the two sources should be used for position at a time, as defined by EK3_SRC_POSXY and EK3_SRC_VELXY. However, if external nav is selected and GPS data is available, the filter incorrectly fuses this data too. This prevents flying on external nav with the ability to switch to GPS as a backup.
It manifests as an unstable position with high variances and resetting behaviour, if the two position estimates diverge - which will be the case with sources like VIO.
The difference can be demonstrated running a replay log with and without GPS data:

[image]
(notice the jagged position and very high innovations with resets)
[image]
(notice the smooth position and much lower innovations - though still a bit high)

Upon inspecting AP_NavEKF3_PosVelFusion.cpp, it seems the if statements in FuseVelPosNED are just not correctly written to separate external nav from GPS. What happens is that FuseVelPosNED is called with `extNavVelToFuse == gpsDataToFuse == fuseVelData == fusePosData == 0`, but with `fuseHgtData == 1` (presumably due to only having baro data). Thus when an if statement like the following is encountered:
```
if (extNavVelToFuse) {
                obs_data_chk = sq(constrain_ftype(extNavVelDelayed.err, 0.05f, 5.0f)) + sq(frontend->extNavVelVarAccScale * accNavMag);
            } else
#endif
            {
                obs_data_chk = sq(constrain_ftype(frontend->_gpsHorizVelNoise, 0.05f, 5.0f)) + sq(frontend->gpsNEVelVarAccScale * accNavMag);
            }
```

The else statement is triggered, incorrectly, incorporating GPS data when in fact external nav is selected. There needs to be an extra check for gpsDataToFuse or something like that.

**Version**
Ardupilot 4.6.0 beta 1 and 2

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
VTOL 

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

**Related Issues**
https://github.com/ArduPilot/ardupilot/issues/25189
https://github.com/ArduPilot/ardupilot/issues/15859
https://github.com/ArduPilot/ardupilot/pull/18380
https://github.com/ArduPilot/ardupilot/issues/23485
https://github.com/ArduPilot/ardupilot/pull/23616

rmackay9, 2025-05-15:
Hi @eufrizz,

Thanks for the report.

A couple of related fix has gone in recently so I wonder if this might resolve the issues you're seeing

- https://github.com/ArduPilot/ardupilot/pull/29420
- https://github.com/ArduPilot/ardupilot/pull/29847

eufrizz, 2025-05-15:
Hi @rmackay9,

Thanks for the heads up, looks like we converged on some of the same fixes there. Still some more fundamental issues that required the fixes added in my PR https://github.com/ArduPilot/ardupilot/pull/30080

Ryanf55, 2025-05-18:
Have you considered the visual odom interface to ArduPilot? Visual Odom has different characteristics than GPS.
https://ardupilot.org/dev/docs/ros2-cartographer-slam.html

eufrizz, 2025-05-19:
@Ryanf55 this was using visual odom. Referring to it as external nav.

rmackay9, 2025-06-26:
I'm pretty sure this has been fixed in master now that PR https://github.com/ArduPilot/ardupilot/pull/29213 has been merged

rmackay9, 2025-12-03:
I'm going to go ahead and close this but I will see responses so we can re-open if I'm wrong, thanks!

closed 2025-12-03 by rmackay9