ArduPilot/ardupilot #30113: SITL: Illegal Instruction on clang

jeffmelville, 2025-05-19:
## Bug report

ArduPlane SITL crashes with ``EXC_BAD_INSTRUCTION`` on Apple Silicon compiled with clang (not sure on version dependency)

Discussion: https://discuss.ardupilot.org/t/bad-instruction-on-sitl-on-macos-osx-m4/134130

```
stop reason = EXC_BAD_INSTRUCTION (code=1, subcode=0x1e201863)
arduplane`Plane::calc_speed_scaler:
<snip>
    0x100004060 <+188>: fmul   s2, s9, s2
    0x100004064 <+192>: fdiv   s2, s3, s2
    0x100004068 <+196>: fmov   s4, #2.00000000
    0x10000406c <+200>: fcmp   s2, s4
    0x100004070 <+204>: fcsel  s2, s4, s2, mi
->  0x100004074 <+208>: fdiv   s3, s3, s0
    0x100004078 <+212>: mov    w8, #0x1de7 ; =7655
    0x10000407c <+216>: movk   w8, #0x3ba7, lsl #16
    0x100004080 <+220>: fmov   s4, w8
    0x100004084 <+224>: fcmp   s0, s4
    0x100004088 <+228>: fcsel  s0, s3, s2, gt
    0x10000408c <+232>: mov    w0, #0x18 ; =24
```

Other trace from the discussion thread:

```
validate_structures:526: Validating structures
Loaded defaults from Documents/GitHub/ardupilot/Tools/autotest/default_params/gazebo-zephyr.parm
warning: arduplane was compiled with optimization - stepping may behave oddly; variables may not be available.
Process 11439 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_INSTRUCTION (code=1, subcode=0x1e201863)
    frame #0: 0x0000000100003f84 arduplane`Plane::calc_speed_scaler(this=0x00000001002eecb8) at Attitude.cpp:19:13 [opt]
   16  	        const float airspeed_min = MAX(aparm.airspeed_min, MIN_AIRSPEED_MIN);
   17  	        const float scale_min = MIN(0.5, g.scaling_speed / (2.0 * aparm.airspeed_max));
   18  	        const float scale_max = MAX(2.0, g.scaling_speed / (0.7 * airspeed_min));
-> 19  	        if (aspeed > 0.0001f) {
   20  	            speed_scaler = g.scaling_speed / aspeed;
   21  	        } else {
   22  	            speed_scaler = scale_max;
Target 0: (arduplane) stopped.
Process 11439 launched: '/Users/ibrahimsefer/Documents/GitHub/ardupilot/build/sitl/bin/arduplane' (arm64)
(lldb) 
```

After some trouble shooting, I'm 90% sure this is the issue: 
https://developer.arm.com/documentation/dui0773/l/Coding-Considerations/Floating-point-division-by-zero-errors-in-C-and-C---code

The ``fdiv`` instruction is getting reordered ahead of the comparison against the threshold based on the assumption that a divide by zero will return ``NaN`` and not fault. ArduPilot calls ``feenableexcept`` that causes the division to generate a CPU exception.

I tried the suggested ``volatile`` fix from the ARM docs and it worked. I can submit that as an MR if there's not a more general fix.

**Version**
Plane 4.6.0-beta1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
Macbook M1 Pro (Apple Silicon) with clang compiler

peterbarker, 2025-05-20:
> I tried the suggested `volatile` fix from the ARM docs and it worked. I can submit that as an MR if there's not a more general fix.

Please do - for discussion at least!

timtuxworth, 2025-05-20:
That would be great if that works, this happens quite often on my Mac. The other case I often is using BATRESET (but not every time).

The workaround for the above error is to set --speedup=1 on the command line. You shouldn't need to set 1, but setting it to 1 explicitly means you don't hit this condition.

khancyr, 2025-05-20:
worth to try if using our is_zero function make the compiler happy ?

peterbarker, 2025-12-10:
We've merged a fix for this.  Please re-open this if problems haven't been fixed.

closed 2025-12-10 by peterbarker