ArduPilot/ardupilot #30142: Bogus Telemetry message about missing RC reciever

Hwurzburg, 2025-05-23:
Plane 4.6, we send occasional FRSKY/CRSF telem warnings about "No RC Receiver" when its obviously there and working from the logs....it does not appear in MAVLink texts to the GCS...only RC telem messages (it only apepars on TX Yapuu display, not in OSD )

peterbarker, 2025-05-23:
"No RC Receiver" is incorrect; this will show if we've seen an RC receiver but it is now unhealthy.

timtuxworth, 2025-05-24:
> "No RC Receiver" is incorrect; this will show if we've seen an RC receiver but it is now unhealthy.

I've seen this many times. I don't know what it means or what should I do about it. Usually I ignore it and everything seems fine, the plane flies, the RC works so I really don't understand what this message is doing for me.

rmackay9, 2025-06-02:
I've added this to the [4.6 issues list](https://github.com/ArduPilot/ardupilot/issues/28612)