ArduPilot/ardupilot #30207: Plane: Q>FW transition does not remove Q_TRIM_PITCH offset in some modes

LzREngineering, 2025-05-31:
## Bug report

**Issue details**
I use the Q_TRIM_PITCH param to set an upward pitch on my VTOL plane in Q-modes, which helps get some 'free' lift in the wind and also gives more control authority for the vectored yaw. In FW modes this offset needs to be removed for level flight; PTCH_TRIM_DEG should be used in those modes to provide an equivalent function if needed (I have this set to 0). However, during certain transitions this does not happen:

- QLOITER>FBWA or FBWB = the Q_TRIM_PITCH offset is removed and the horizon uses PITCH_TRIM_DEG as an offset.

- QHOVER, QSTABILIZE>FBWA or FBWB = the Q_TRIM_PITCH offset does **not** change when the fixed wing mode is selected, generally resulting in a crash.

I would like to be able to build up speed in QSTABILIZE then switch to FBWA for a more 'manual' transition, however this currently won't work if the pitch offset does not change as it should.

Note, this also seemed to exist in beta versions of 4.6.0 but seems to still be present in the official release.

**Version**
4.6.0 official

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ / ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tilt-rotor VTOL (4 tilting motors)

**Hardware type**
Matek h743 v3

**Logs**
not yet - only checked on ground

IamPete1, 2025-05-31:
Can you upload a dataflash log of your failed test please.

LzREngineering, 2025-05-31:
[image]

The symptoms are actually slightly different when the flight mode is changed via RC: QHOVER>FBWA does work correctly in this case (previously I was changing flight modes in Mission Planner). The QSTABILIZE>FBWA change consistently does not work correctly.

log: [https://w-si.link/YtLaZAKPaa7Zipc28](https://w-si.link/YtLaZAKPaa7Zipc28)

IamPete1, 2025-06-01:
This is due to the controller that is used for the first stage of the transition.

https://github.com/ArduPilot/ardupilot/blob/baab8528d60baaeabd25c3ff2eaceb7c04a11ca2/ArduPlane/quadplane.cpp#L4271-L4279

There is no fundamental difference between modes, or the source of the mode change. It is due to being in throttle wait here:

https://github.com/ArduPilot/ardupilot/blob/baab8528d60baaeabd25c3ff2eaceb7c04a11ca2/ArduPlane/quadplane.cpp#L1673-L1681

If throttle wait is active transition is skipped so the FW controller is used. If throttle wait is not active (like when your really flying) then the transition goes into the airspeed wait phase. Since you never reach airspeed this never completes.

So, the code does seem to be working as it should. The goal of using the VTOL controllers for the first stage is to prevent any vector miss-alignments from causing large yaw changes. The FW controller has no vectored yaw control to prevent this.

I can see why the large pitch offset might cause issues in this case, we could look into ways to mitigate. It will clear once the transition is completed.