ArduPilot/ardupilot #30208: Copter: Build failure for SpeedyBee F4 V3 on ArduPilot v4.6.0 due to linker error

AntonChaukin, 2025-05-31:
I'm encountering a build issue when compiling ardupilot v4.6.0 for the SpeedyBee F4 V3 flight controller.

```
/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld:common.ld:202 cannot move location counter backwards (from 0000000008102860 to 0000000008100000)
collect2: error: ld returned 1 exit status

Waf: Leaving directory `../git/ardupilot_f/vtx/speedybeef4v3'
Build failed
 -> task in 'bin/arducopter' failed (exit status 1)

```

**Issue details**

I run commands:
```
./waf configure --board speedybeef4v3 -o vtx
./waf copter
```

This issue does not occur on version v4.4.3 — the build completes successfully there.
I’ve already tried reducing the binary size by disabling unnecessary modules, both in:

`libraries/AP_HAL_ChibiOS/hwdef/speedybeef4v3/hwdef.dat`

`Tools/scripts/build_options.py`

The result remains the same, and I still get the linker error.

**Version**
ardupilot v4.6.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2025-05-31:
Please make sure your using the correct compiler, your commands work for me on `g++ 10.2.1`.

AntonChaukin, 2025-06-01:
My g++ version is 10.4.0
I used `Tools/environment_install/install-prereqs-ubuntu.sh -y` before compiling

AntonChaukin, 2025-06-01:
I also tried building using docker — the same linker error occurs

IamPete1, 2025-06-01:
The error means you ran out of flash. If your using the correct compiler you must have some custom changes that cost use more flash than the standard build. Can you share the exact branch?

AntonChaukin, 2025-06-01:
hah

I made a new dir, cloned ardupilot, next:
`git fetch --all --tags`
`git checkout tags/Copter-4.6.0`
`Tools/environment_install/install-prereqs-ubuntu.sh -y`
`. ~/.profile`
`./waf configure --board speedybeef4v3 -o vtx`
`./waf copter`
and it works!

but, if I want to enable scripting -  the same linker error occurs

can I compile build for speedybeef4v3 with scripting?

IamPete1, 2025-06-01:
> can I compile build for speedybeef4v3 with scripting?

Possibly, you need about 150K free, you will have to turn off other stuff until it fits. 

In anycase, I think the original issues is resolved.

closed 2025-06-01 by IamPete1