ArduPilot/ardupilot #30242: GCS_SEND_TEXT messages should support different languages

rmackay9, 2025-06-05:
AP sends many important messages to the ground station and we should support displaying these in different languages.  It's not only convenient but especially for important warnings it can be a safety issue

This request was received at the 2025 Japan Drone Expo

timtuxworth, 2025-06-05:
I have had many requests for this from China, Indonesia and Brazil (at least), including from several partners.

timtuxworth, 2025-06-05:
> [#17244](https://github.com/ArduPilot/ardupilot/pull/17244) is relevant.
> 
> You can't really "translate" `GCS_SEND_TEXT` messages as different languages want different substitution strings in different orders - so your string ("Your %s %d is > %u") may need to be `XYZZY %d YY >%u %s` in a different language.
> 
> The mavlink events interface is supposed to allow for this sort of thing.

I think the issue as written by Randy doesn't mention GCS_SEND_TEXT. Maybe events are an option, maybe doing something with send text, maybe some other way, but language is a UI thing which should be done on the GCS, not on the AP.