ArduPilot/ardupilot #30258: Feature request: Add auxiliary functions for Zoom and Focus absolute control

TheRekiker, 2025-06-06:
## Feature request: Add auxiliary functions for Zoom and Focus absolute control

Right now exist 2 auxiliary functions that allow relative control of camera zoom (Auxiliary function 167) and focus (Auxiliary function 168). This functions only allow inputs as High, Mid or Low which don’t allow for absolute control. It could be a nice feature to have 2 aditional Auxiliary functions that allow PWM input and absolute control of Zoom and Focus.

**Platform**

[X] Copter
[X] Plane
[X] Rover
[X] Submarine

rmackay9, 2025-06-07:
I think this is a good suggestion.  I think ideally it should be the ground stations (QGC, MP) that add better controls to their existing custom gimbal controls but we could also add RC controls (which is what this issue is requesting)

olliw42, 2025-06-07:
just to correct/clarify. STorM32 is not expecting CAMERA_ZOOM_TYPE = ZOOM_TYPE_RANGE (2), it in fact supports ZOOM_TYPE_STEP (0) and ZOOM_TYPE_RANGE (2).
(it does so however only for a GoPro camera, and not for the VISCA camera @TheRekiker is using, but that's not AP's cup of tea).