ArduPilot/ardupilot #30284: Harmonic Notch Filter Bandwidth greater than 2 x center frequency is not constrained

bnsgeyer, 2025-06-09:
## Bug report

**Issue details**

A bandwidth greater than twice the center frequency was accidentally used which resulted in large lags in the filtered response making the aircraft nearly unflyable.  it was easily reproducible in SITL.  It was present not only in 4.6.0 but also 4.5.7

**Version**
4.5.7 and 4.6.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter - Heli
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Heli

**Hardware type**
This was first encountered on a Cube orange plus in real life.  Data log provided was conducted in SITL for heli frame type

**Logs**
The attached log shows the issue with heli in SITL on Copter 4.6.0.   The INS_GYRO_RATE was set to 1 and the INS_FAST_SAMPLE was set to 7 as this were the setting on the heli which had the issue in real life.  In this aircraft, center frequency is set to 9 hz and the bandwidth set to 40hz which produced the lagged response.  It is most notable on the IMU0.gz compared to IMU1.gz shown here.

[image]

link to log is here
https://drive.google.com/file/d/1v0X5433Ez8ADqn1HGYNyYcjJrkJzvDyv/view?usp=sharing

lthall, 2025-06-09:
This should stop the notch bandwidth from being more than 2 x centre frequency.

https://github.com/ArduPilot/ardupilot/blob/211157c38b425de2cf018cab65e6cafdbd445483/libraries/Filter/NotchFilter.cpp#L34

bnsgeyer, 2025-06-09:
> INS_HNTCH_REF is 0 and INS_HNTCH_MODE is 1 for throttle. Throttle based makes no sense on heli, even if it did then ref of 0 means it will fall back to static. You should have hit the new pre-arm for using throttle based on heli (maybe its not in 4.6).

@IamPete1 No doubt that the user had some bad settings for a heli.  I verified that this is still an issue with proper settings being INS_HNTCH_MODE 2 and INS_HNTCH_REF 1.

andyp1per, 2025-06-10:
> This should stop the notch bandwidth from being more than 2 x centre frequency.
> 
> [ardupilot/libraries/Filter/NotchFilter.cpp](https://github.com/ArduPilot/ardupilot/blob/211157c38b425de2cf018cab65e6cafdbd445483/libraries/Filter/NotchFilter.cpp#L34)
> 
> Line 34 in [211157c](/ArduPilot/ardupilot/commit/211157c38b425de2cf018cab65e6cafdbd445483)
>  if (center_freq_hz > 0.5 * bandwidth_hz) {

Right, it should - I am not sure why this would be happening