ArduPilot/ardupilot #30292: Upgrade numpy and other Python dependencies

cclauss, 2025-06-09:
@peterbarker >  We have many warnings coming out of our tools saying we're using deprecated libraries and APIs; eg. we're restricted to numpy < 2 in some of our tooling, and that's not playing nice with tools created this century. If I got to point people at fixing problems with our Python code, that is definitely what I would suggest!
* https://github.com/ArduPilot/ardupilot/pull/30278#pullrequestreview-2908664909

I could not find any open issues about Numpy.  I could not find where Python dependency versions were being pinned.  For example, [searching for `numpy`](https://github.com/search?q=repo%3AArduPilot%2Fardupilot%20numpy&type=code) in this repo only libraries/AP_NavEKF3/derivation/generate_nix and Tools/environment_install/install-prereqs-arch.sh seem to be selecting a numpy version.  The former via a Nix package hash and the latter via apt package.  Both must be out of date if they are still installing Numpy < v2 a year after it was released (Numpy v2.3 is current).  The usual `pyproject.toml`, `requirements.txt`, `setup.cfg`, `setup.py`, or PEP723 headers methods of specifying Python dependencies are not used in this repo.  Is some other Python dependency that is preventing us from upgrading to Numpy v2?

## Feature request

**Is your feature request related to a problem? Please describe.**
Understand how and where Python dependencies are being pinned in this repo on various operating systems.
Understand why Python dependencies are being pinned to out-of-date versions.
Enable them to be upgraded regardless of operating system.

**Describe the solution you'd like**
A clearly defined way to pin and upgrade Python dependencies such as `pyproject.toml` or PEP723 headers.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[x] All

**Additional context**
_Add any other context or screenshots about the feature request here._

tpwrules, 2025-06-09:
The `AP_NavEKF3` mustn't be changed, it's pinned as that's the version that generates the EKF code in tree. We need to do a bunch of EKF testing if we want to change it.

cclauss, 2025-06-15:
@peterbarker wrote https://github.com/ArduPilot/ardupilot/pull/30361#issuecomment-2973677097
> Just a note that upgrading numpy won't necessarily kill in-tree tools, but most definitely kills MAVProxy, which is usually run in the same venv. This does suck.

---

I do not see anything that restricts the version of NumPy used in MAVProxy.
* https://github.com/search?q=repo%3AArduPilot%2FMAVProxy%20numpy&type=code

The `pysim` issues mentioned above restrict the version of NumPy, but MAVProxy does not use `pysim`.

In this repo, is `pysim` a runtime dependency or only a test-time (autotest-time ;-]) dependency?

% `git grep --name-only pysim`
```
Tools/autotest/arducopter.py
Tools/autotest/arduplane.py
Tools/autotest/autotest.py
Tools/autotest/examples.py
Tools/autotest/helicopter.py
Tools/autotest/pysim/__init__.py
Tools/autotest/pysim/util.py
Tools/autotest/rover.py
Tools/autotest/run_mission.py
Tools/autotest/sim_vehicle.py
Tools/autotest/test_build_options.py
Tools/autotest/test_param_upgrade.py
Tools/autotest/vehicle_test_suite.py
Tools/scripts/build_examples.py
Tools/scripts/param_check_all.py
Tools/scripts/sitl-on-hardware/sitl-on-hw.py
```