ArduPilot/ardupilot #30356: Overflow Issues in MAV Commands

wyunc, 2025-06-14:
## Bug report

Hi : )
I previously reported some of these on discuss.ap to Lucas(amilcarlucas). For easier tracking, I’d like to summarize here a few MAV commands that have been found to cause overflow issues.

The following commands can trigger arithmetic overflow or floating-point overflow when given certain special values. In most cases, the cause is that the x, y, or z values are not properly handled or constrained.

1.The following are present in both copter,plane, and rover:
mavcmd MAV_CMD_DO_SET_HOME [1, 180, 84, 180, 399, 300, 699]
mavcmd MAV_CMD_DO_SET_ROI [1, 37, 479, 0, 706, 608, 441]
mavcmd MAV_CMD_DO_SET_ROI_LOCATION [426, 0, 0, 0, -1381736522, -585139360, 135]
mavcmd MAV_CMD_CAN_FORWARD [498, 0, 0, 0, 0, 0, 0]
mavcmd MAV_CMD_EXTERNAL_POSITION_ESTIMATE [25, 23, 434, 0, 834, 25, 916]

2.The following command only works with copter:
mavcmd MAV_CMD_NAV_LOITER_TO_ALT [1, 650, 0, 0, 458, 291, 615]

**Version**
4.7.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover
[  ] Submarine

peterbarker, 2025-07-07:
We are quite aware that we're not bounds-checking stuff coming in over mavlink.  For extra fun try passing in NaN values in the parameters.

We are very flash-contrained on our embedded targets.  Sanity checking every parameter is *costly* in terms of flash.

If you find bug where we're not simply doing bad things in terms of types (eg. buffer overflows), we will definitely care.

... and we will add sanity checks - this PR of mine, for example: https://github.com/ArduPilot/ardupilot/pull/28866/files

wyunc, 2025-07-21:
Hi @peterbarker ,thanks for your response and the clarification.

Apologies for the delayed reply — I’ve been swamped with a deadline recently...

I fully understand the flash constraints on embedded targets, and I agree that checking every incoming parameter is not always feasible. That said, the inputs I reported are not just about unbounded values or random edge cases — they consistently trigger arithmetic(floating-point) overflow during execution. 

I also noticed that the issues I encountered are very similar to those described in #28790, especially in how unexpected MAVLink input leads to unsafe or undefined behavior. So while I’m not suggesting widespread input validation, I do think these particular cases represent valid bugs, as they result in observable runtime errors in normal test scenarios.

Could you help confirm whether these overflow issues should be tracked as bugs (or enhancements), and whether the team would consider adding minimal bounds checks or clamping for these specific commands?