ArduPilot/ardupilot #30373: Motor test

StenPetrov, 2025-06-16:
## Bug report

Motor test doesn't take into account reversible motor settings, spins ALL motors near max speeds

**Issue details**

MOT_PWM_TYPE = 6 (Dshot600)
SERVO_DSHOT_ESC=3 (AM32+EDT)
SERVO_BLH_3DMASK=15 (first 4 channels, where the motors are attached)
SERVO_BLH_RVMASK=15

Go to MissionPlanner's Motor Test, set Throttle%=3, click "Test motor A"

**Expected: one motor spins at low RPM.**
**Actual: ALL motors spin near max RPM**

The issue exists regardless of 3D mode setting on the ESCs, as configured from am32 configurator.

**Version**
 ChibiOS: 88b84600
 ArduCopter V4.6.0-dev (a2be43fc) - stock FW from manufacturer

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
 TBS_LUCID_H7 

**Logs**
N/A

peterbarker, 2025-07-07:
ANy chance of a dataflash log file of this happening?

peterbarker, 2026-01-12:
> [@peterbarker](https://github.com/peterbarker) There are a bunch of log files, one of them should have the motor test, but I don't know how to pick the right one. Is there some specific event or anything that would indicate a motor test was run?

statustext messages, I think.  `CMDI` will contain information about the command used to start the motor test.