ArduPilot/ardupilot #30395: EKF Lane Change Issue

Photronix, 2025-06-18:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

EKF land switch leading to crash

_Please describe the problem_

Since moving to 4.5.1 we have seen a high occurrence of EK3 lane switch.  Most of the time this results in uninterrupted flight.  However, there have been several (3 or more) instances of crashes due to what we see as the choosing of a bad lane.

Auto analysis yields the following:

Size (kb) 10933.373046875
No of lines 129497
Duration 0:03:24
Vehicletype ArduCopter
Firmware Version V4.5.1
Firmware Hash
Hardware Type
Free Mem 0
Skipped Lines 0
Test: Autotune = UNKNOWN - No ATUN log data
Test: Brownout = GOOD -
Test: Compass = FAIL - FAIL: Large compass offset params (X:162.81, Y:357.25, Z:-503.16)
FAIL: Large compass offset in MAG data (X:-389.00, Y:416.00, Z:1360.00)
Large change in mag_field (1970.55%)
Max mag field length (5074.91) > recommended (550.00)
Test: Dupe Log Data = GOOD -
Test: Empty = GOOD -
Test: Event/Failsafe = FAIL - ERRs found: FLT_MODE CRASH GPS_GLITCH
Test: GPS = UNKNOWN - join() takes exactly one argument (2 given)
Test: IMU Mismatch = NA -
Test: Motor Balance = UNKNOWN - 'QUAD/H'
Test: NaNs = FAIL - Found NaN in CTUN.TAlt
Found NaN in CTUN.DSAlt
Test: OpticalFlow = FAIL - 'FLOW_FXSCALER' not found
Test: Parameters = FAIL - 'MAG_ENABLE' not found
Test: PM = UNKNOWN - 'NLoop'
Test: Pitch/Roll = UNKNOWN - 'BarAlt'
Test: Thrust = GOOD -
Test: VCC = UNKNOWN - No CURR log data

We would like to understand how parameters EK3_AFFINITY, EK3_ERR_THRESH, mag selection (we can only use the external mag) and others can mitigate this issue.

The attached log and auto analysis suggest it was a magnetometer issue but we have seen other flights where it seems like the lane switch is made to a obviously wrong lane.

**Version**
4.5.1

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
H Frame

**Hardware type**
Cube Orange

**Logs**
https://drive.google.com/file/d/16Wi0zAejf9fTK-DtRrv1ams-8uQEHgqi/view?usp=sharing

rishabsingh3003, 2025-06-22:
The log you have attached does not seem to open up (at least in MAVExplorer). Perhaps download the log again and upload it?

Photronix, 2025-06-23:
I needed to GPS anonymize it...it does open using Mission Planner review log.  Not sure about MAVExplorer but UAV Log Viewer also has trouble opening the log.  If it's a problem I will seek permission to post original.