ArduPilot/ardupilot #30442: thr_slwrate parameters

arhmar, 2025-06-23:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

thr_slwrate parameter does not work

4.6.0
_What version was the issue encountered with_

plan

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2025-06-23:
We need some more info. What did you test? What were you expecting to happen? What actually happened? Did you try anything else? A .bin log from the SD card would help us work out what if going on.

arhmar, 2025-06-29:
It does not affect the throttle servo in FBWA mode, it only works in auto mode.

peterbarker, 2025-07-07:
... OK, so I think this one is explained; the parameter doesn't apply in the situation the user was expecting.

Closing this one now.

closed 2025-07-07 by peterbarker

arhmar, 2025-07-20:
ok thank you