ArduPilot/ardupilot #30474: Foxeer F405V2 - Config Error: Baro: unable to initialise driver

EasyRider-one, 2025-06-25:
just received my F405 V2 Stack (ordered from a French online shop), and I’m trying to use it with ArduPilot. However, I’m running into an issue I haven’t been able to resolve: the barometer is not detected — I get the error message: "Baro: Unable to initialize driver".

Interestingly, when I flash Betaflight, the barometer works perfectly fine.
I’ve tried flashing multiple versions of ArduPilot firmware, but none of them have solved the problem.

I also contacted Foxeer, and they sent me a firmware file (.hex), but even after flashing it, the barometer still doesn’t work under ArduPilot.

Have you ever run into this issue — either with this board or a similar one?
Any advice you could offer to help solve the problem would be greatly appreciated.

Thanks in advance!

EasyRider-one, 2025-06-25:
Okay,
The barometer is a DPS310. If you want more information, here's a link:
[https://www.foxeer.com/foxeer-f405-v2-plug-fc-reaper-55a-esc-8s-stack-video-switcher-servo-borameter-g-578](url)

Thanks

EasyRider-one, 2025-06-26:
Hi,

Thanks for the help. I used devop to test the I²C bus — the barometer responds at 0x76

devop read i2c dps310 0 0x76 0x0 1
Operation 1 OK: 1 bytes
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
   0: 00 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

I’m not very familiar with MAVProxy — just using it for debugging. Let me know if I should try anything else.

Thanks!

andyp1per, 2025-06-26:
> Nice, you solved my issue — thank you so much! I should have tried this earlier, sorry.
> 
> Just to understand better: Was the problem that the barometer is not actually a DPS310, but a SPA06 instead? Did Foxeer change the sensor without updating their documentation or website?
> 
> Thanks again for your support!

Yes and yes.