ArduPilot/ardupilot #30477: Ubuntu Setup: Prereq script exits after path error

avenstewart, 2025-06-25:
## Bug report

**Issue details**

_When the script attempts to setup the environment at the end of the prerequisites installation process on Ubuntu, the script will fail to set $SCRIPT_DIR and $ARDUPILOT_ROOT if the name of the parent folder containing the Ardupilot repository contains a space character, e.g.:

"../Drone Projects/ardupilot/Tools..."

The following error occurs and causes the script to exit prior to completion:

+ echo '##############################################'
##############################################
+ echo Adding ArduPilot Tools to environment
Adding ArduPilot Tools to environment
+ echo '##############################################'
##############################################
+++ realpath Tools/environment_install/install-prereqs-ubuntu.sh
++ dirname /home/agstewart/Desktop/Drone Projects/ardupilot/Tools/environment_install/install-prereqs-ubuntu.sh
+ SCRIPT_DIR='/home/agstewart/Desktop
Projects/ardupilot/Tools/environment_install'
++ realpath '/home/agstewart/Desktop
Projects/ardupilot/Tools/environment_install/../../'
realpath: '/home/agstewart/Desktop'$'\n''Projects/ardupilot/Tools/environment_install/../../': No such file or directory
+ ARDUPILOT_ROOT=
agstewart@dev:~/Desktop/Drone Projects/ardupilot$
 
The problem lies at line 512 in install-prereqs-ubuntu.sh:

SCRIPT_DIR=$(dirname $(realpath ${BASH_SOURCE[0]}))

Wherein the variable substitution should be double quoted to prevent splitting_

**Version**
_built from source on master branch_

**Platform**
[ x] All

avenstewart, 2025-06-25:
Obvious solution is to rename the folder, but I figured that this is still a problem so it should be reported.

avenstewart, 2025-07-19:
@peterbarker @The13bit Still appears to encounter the issue with whitespace specifically only if the user opts to install the Ardupilot Bash completion tools:

`Add ArduPilot Bash Completion to your bash shell [N/y]?y
+ [[ y =~ ^[Yy]$ ]]
+ return 0
+ echo source /home/avenstewart/Desktop/Drone Projects/ardupilot/Tools/completion/completion.bash
+ eval source /home/avenstewart/Desktop/Drone Projects/ardupilot/Tools/completion/completion.bash
++ source /home/avenstewart/Desktop/Drone Projects/ardupilot/Tools/completion/completion.bash
/home/avenstewart/Desktop/Drone Projects/ardupilot/Tools/environment_install/install-prereqs-ubuntu.sh: line 542: /home/avenstewart/Desktop/Drone: No such file or directory
`
I noticed the user answered "n" in the log posted with the fix PR, so I tried testing both, and the issue now only occurs when user opts to install them.

avenstewart, 2025-07-20:
**UPDATE - please read**: This bug will cause Bash to crash on login when trying to initialize the modified .profile for the affected user, leading to the user being unable to authenticate. The whitespace will cause a malformed PATH entry and Bash will exit, followed by GDM kicking the user back to login with a generic "Login incorrect" (from CLI) and simply failing to login without any error from the GUI. 

This means that the user won't be able to simply remove the whitespace from the directory name and reinstall unless they clean up the .profile first before they restart or log out.

Just for documentation sake, here's the log showing the PATH error (the whitespace splits the filepath between "/Drone Projects/"):

```
/home/avenstewart/.profile: line 37: export: `Projects/ardupilot/Tools/autotest:/usr/bin:…
/usr/bin/gnome-session: 6: grep: not found
bash[4123]: Unknown option -l
```
Removing the broken PATH from .profile and removing the whitespace from any ardupilot parent directories fixes this issue, for anyone experiencing it.

peterbarker, 2025-08-12:
@avenstewart looks like you fixed this - OK to close?

closed 2025-08-12 by avenstewart