ArduPilot/ardupilot #30500: PWM Out Duty Glitch even in Direct Passtrough

apxaerospace, 2025-06-27:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

I found 2 problems releated to the RCOUT behaviour. Help requested tru Discord without success, users suggest to open a bug report.

- Speed of the servo that is directly connected to the CH1/2 of the RX is much faster than the servo connected to the FC in manual mode, and much smoother. Video for comparison. Just use an Sbus Rx and connect 1 servo directly to the Rx and one directly tru the FC
- The servo connected to the FC in manual mode looks a bad stepper motor with huge vibrations and steepy behaviour. Large fiber parts might amplificate the noise

I suppose that this behaviour might lead to vibrations that are not normally produced by the structure and motors, but from the servo itself.

**TEST DONE**

[image]

- Changed FW version from 4.6.0 to 4.4.0 = FAIL to little or no changes
- Change servo with analog and another digital one (total 4 different servos) = FAIL
- Change servo rate from 50 to 333Hz = FAIL
- Oscilloscope check on servo signal that comes out from FC, no EMI = PASS
- PWM voltage more than 3,3V = PASS
- FBWA Mode = FAIL, nothing changes
- Manual Mode
- Direct Passtrough from channel 1 to RCOUT = FAIL

**Version**
tested 4.4.0 and 4.6.0, not yet tested newer and older version, but experiments is on going

**Platform**

https://github.com/user-attachments/assets/b3cbdf4f-a51e-4deb-b22c-8db6720a79f3
https://github.com/user-attachments/assets/dd436622-0add-4c7e-bf50-77499586c328
[image]

Arduplane

**Airframe type**
Flying Wing

**Hardware type**
CUAV V6X + HobbyWing BEC + Power Monitor CANbus + 2x Savox Servo SB2271SG + 12S power supply + Futaba R7008SB

**Logs**
1x Video that shown servo directly connected to the CH1/2 of the RX
1x Video that shown the same servo connected to the FC in Manual Mode
1x Picture that shown small glitch in CHout

Logs available, other videos available

apxaerospace, 2025-06-27:
[log_files.zip](https://github.com/user-attachments/files/20957075/log_files.zip)

ahmedmsalah99, 2025-07-30:
Hello @apxaerospace 
I was curios about your problem and I was thinking what might have caused it. 

The only thing that I can think of is the drop in voltage (as shown in the screenshot) when PWM gets out of the FC. 

It looks like Savox Servos are power hungry which means they maybe not having enough energy from the PWM coming out from the FC compared to the pure energy directly from the RX.

Having your setup with load on the servos makes this even more probable.