ArduPilot/ardupilot #30528: Filter Review Tool Broken Button

magate, 2025-07-02:
## Bug report

**Issue details**

[image]

Seems to have a broken button at the bottom of the filter review tool webpage. Tested in Chrome.

**Version**

Webtools

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Webtools

**Hardware type**

Webtools
**Logs**

Webtools

IamPete1, 2025-07-02:
Fixed, should be live within the next 30 mins or so.

Note that I would not recommend use the old filter tool anymore, it has not been kept up-to date with the latest filtering changes. The filter review bode plot should be  more accurate to what AP does. Is there some reason why you prefer the original? Maybe we could re-work the plot on the review tool to work better for you.

closed 2025-07-02 by IamPete1

IamPete1, 2025-07-02:
> [@IamPete1](https://github.com/IamPete1) is this the one you are talking about: https://firmware.ardupilot.org/Tools/WebTools/FilterTool/? I didn't realize there was another one.

Correct. The filter review tool does try and give a bode estimate too (the plot above the open in filter tool button).