ArduPilot/ardupilot #30580: Auto focus not working for siyi camera

jjmmss00, 2025-07-07:
## Bug report

**Issue details**  Commanding autofocus in lua  or via rc command scrip guves no result

_Please describe the problem_
 inspecting ihe c++ file AP_Mount_Siyi.cpp, the auto focus command is called with no parameters, while the siyi interface document requires an x and y coordinate (in pixels) for focusing point.

**Version**
_What version was the issue encountered with_
latest - updated several days ago
**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_  
quad copter

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_ 
pixhawk 6x plus SIYI ZR30 camera

**Logs**
_Please provide a link to any relevant logs that show the issue_
no logs. observed in camera video plus from inspecting ardupilot code.

jjmmss00, 2025-07-09:
I can not detect any change in my camera focus when I issue this command ether through an RC command or through a lua script. Change in focus is very obvious with the working Manuel focus command......it is interesting from your comment that SIYI has changed their driver interface. I was not aware of that.

rmackay9, 2025-07-09:
Hi @jjmmss00,

Thanks very much for the testing and report.  I'll try and fix this although I have quite a few things on my plate at the moment

khanasif786, 2025-07-12:
@rmackay9 are you working on it, cause i think i may fix this. let me know.

rmackay9, 2025-07-12:
Hi @khanasif786, I haven't started on it yet so it's all yours!

khanasif786, 2025-07-13:
@jjmmss00 can you test it, unfortunately , lately realised i dont have access to SiYi camera right now

jjmmss00, 2025-07-14:
OK. How do I get access to updated firmware?

khanasif786, 2025-07-14:
oh this PR is not merged yet let me give you build for it to test, i will give .bin file that you can test. just wait a sec.

khanasif786, 2025-07-14:
See you can wait for [these](https://github.com/ArduPilot/ardupilot/pull/30630/checks) to be completed successfully and then download a build, what board are you using ?

khanasif786, 2025-07-14:
meanwhile you can try this, keep in mind some saftey workflows: 
- for safety don't fly it during test cause its just a gimbal patch so it can be tested on ground
- its a custom build so only use it with quads propellers removed (or confirm it will not cause any problem).
- take your parameter lists backup somewhere just i case its lost.

[arducopter.zip](https://github.com/user-attachments/files/21221190/arducopter.zip)

jjmmss00, 2025-07-14:
I tried the new file, but am having some trouble with cammera having just removed it from drone and reinstalled it. I will test the fix just as soon as I get things working again. Pleases give me a day.

jjmmss00, 2025-07-15:
I have not been able to test new release. My camera has developed an issue where it automatically goes to full zoom when powered on, and may need some repair. I will get back to you if I can get it resolved.