ArduPilot/ardupilot #30826: Sagetech MXS ADSB Broken

mareschn, 2025-08-04:
## Bug report

**Issue details**

It appears that we can't use the Sagetech MXS ADSB with Ardupilot due to impossible altitude data provided by the autopilot; the altitude data fields are all zeros. One of those zero bits is telling the ADSB that the autopilot is unable to provide altitude, and another tells the ADSB that it must not use the unit's internal pressure sensor; therefore, it breaks without any source of altitude available.

**Version**
V4.6.2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
Cube Orange

**Logs**
In capturing some serial data from the autopilot to the MXS ADSB, I see operating messages that look like this:
AA 03 02 0C 02 80 02 00 **00 00** FF F9 EB D1 80 01 74

AA=Start Byte
03=Operating Message
02=Incrementing Sequence ID
0C=Payload Length (12 Bytes)
02=Payload Byte 0
80=Payload Byte 1
02=Payload Byte 2
00=Payload Byte 3
**00=Payload Byte 4
00=Payload Byte 5**
ETC.

I'm not sure if there are other problems, but payload bytes 4 and 5 (Altitude) are what I noticed cause trouble: 
Bits 0-13 are Altitude data starting at -1200 ft with either 25 or 100ft scaling (so it is indicating -1200ft in either scaling case)
Bit 14 is the Host [Ardupilot] Altitude Available bit (0=unavailable, 1=available)
Bit 15 is the Internal [ADSB] Altitude Enable bit (0=do not use internal sensor altitude, 1=use internal sensor altitude)

It should be somewhat easy to provide altitude to the ADSB, but if not, bits 14 and 15 should not both be zero.

If the autopilot can't provide altitude, then it should set bit 15 to allow the use of the internal altimeter (0x8000). It is not possible to use the MXS with ArduPilot in its current state because there is no way of getting any altitude data.

magicrub, 2025-08-04:
@faberc are you able to look into this?

faberc, 2025-08-04:
Unfortunately I no longer work with Sagetech and thus don't have access to their IP or hardware to do any testing with so my ability to help here is limited.

I recommend reaching out to support@sagetech.com.