ArduPilot/ardupilot #30838: 架构错误

lyh156, 2025-08-06:
在树莓派4b(arm64/aarch64)使用源码搭建ArduPilot固件编译环境使用该命令Tools/environment_install/install-prereqs-ubuntu.sh -y，后续出现如下错误

[image]

peterbarker, 2025-08-06:
Check your submodules.

lyh156, 2025-08-06:
> 检查您的子模块。

检查和更新子模块的方法都尝试过，错误依然存在，

peterbarker, 2025-08-06:
> > 检查您的子模块。
> 
> 检查和更新子模块的方法都尝试过，错误依然存在，

Please drop in the translated-to-English versions (or just the text versions) of those error messages (rather than an image).  Unfortunately I only read English!

peterbarker, 2025-08-06:
> > > > 检查您的子模块。
> > > 
> > > 
> > > 检查和更新子模块的方法都尝试过，错误依然存在，
> > 
> > 
> > 请输入这些错误消息的翻译成英文版本（或仅文本版本）（而不是图像）。不幸的是，我只读英语！
> > Both methods of checking and updating submodules have been tried, but the error still exists.

I was referring to the compilation errors present in your original image.

lyh156, 2025-08-06:
> 如果你往里看，你看到这样的东西吗？`modules/CrashDebug`
> 
> ```
> pbarker@crun:~/rc/ardupilot(master)$ ls modules/CrashDebug/
> bins          include    libCrashDebug  LICENSE  makefile  README.creole
> CrashCatcher  libcommon  libmocks       main     mri
> pbarker@crun:~/rc/ardupilot(master)$ 
> ```
> 
> 其次，请回顾整个编译 - 子模块在编译过程中可能出现错误，这是一

> 如果你往里看，你看到这样的东西吗？`modules/CrashDebug`
> 
> ```
> pbarker@crun:~/rc/ardupilot(master)$ ls modules/CrashDebug/
> bins          include    libCrashDebug  LICENSE  makefile  README.creole
> CrashCatcher  libcommon  libmocks       main     mri
> pbarker@crun:~/rc/ardupilot(master)$ 
> ```
> 
> 其次，请回顾整个编译 - 子模块在编译过程中可能出现错误，这是一个级联故障。
The issue has been resolved, thanks for your help