ArduPilot/ardupilot #30850: [mavproxy.py -3] AP: DDS: No ping response, exiting

adityapruthi01, 2025-08-07:
## Bug report

**Issue details**

When I run this commands:
`source /opt/ros/humble/setup.bash
cd ~/ardu_ws/
colcon build --packages-up-to ardupilot_sitl
source install/setup.bash
ros2 launch ardupilot_sitl sitl_dds_udp.launch.py transport:=udp4 refs:=$(ros2 pkg prefix ardupilot_sitl)/share/ardupilot_sitl/config/dds_xrce_profile.xml synthetic_clock:=True wipe:=False model:=quad speedup:=1 slave:=0 instance:=0 defaults:=$(ros2 pkg prefix ardupilot_sitl)/share/ardupilot_sitl/config/default_params/copter.parm,$(ros2 pkg prefix ardupilot_sitl)/share/ardupilot_sitl/config/default_params/dds_udp.parm sim_address:=127.0.0.1 master:=tcp:127.0.0.1:5760 sitl:=127.0.0.1:5501`

There comes an error describing 
DDS: No ping response, exiting
In the Mavproxy Instance when i try to run it through
`mavproxy.py --console --map --aircraft test --master=:14550`
I am following this page: https://ardupilot.org/dev/docs/ros2-sitl.html

**Version**
Arducopter 4.5

**Platform**
[  ] Copter

**Airframe type**
SITL

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**

[image]

peterbarker, 2025-08-13:
@adityapruthi01 please try with 4.6 instead.

@Ryanf55 can you tell at a glance what might have gone wrong here?

peterbarker, 2025-10-19:
> Thanks for your reply. Is there any specific command to install 4.6? I have just followed this https://ardupilot.org/dev/docs/building-the-code.html#setting-up-the-build-environment
> 
> Kindly revert.

`git checkout Copter-4.6.2` will get you the 4.6.2 release of Copter, for example.

adityapruthi01, 2026-02-12:
Hi peter yes I am still having issues in this. I even can see your name in my commands. 
ROSbuntu:~/ardu_ws/src/ardupilot$ cd modules/mavlink
git log -1
commit 0c93d60071037c9d0f66bbbbd8f72b564e196705 (HEAD)
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Tue Feb 10 13:05:44 2026 +1100

    common.xml: copy MAV_FTP_ERR and MAV_FTP_OPCODE from mavlink/mavlink/master

sickle@ROSbuntu:~/ardu_ws$ ros2 launch ardupilot_sitl sitl_dds_udp.launch.py \
transport:=udp4 \
synthetic_clock:=True \
wipe:=False \
model:=quad \
speedup:=1 \
slave:=0 \
instance:=0 \
defaults:=$(ros2 pkg prefix ardupilot_sitl)/share/ardupilot_sitl/config/default_params/copter.parm,$(ros2 pkg prefix ardupilot_sitl)/share/ardupilot_sitl/config/default_params/dds_udp.parm \
sim_address:=127.0.0.1 \
master:=tcp:127.0.0.1:5760 \
sitl:=127.0.0.1:5501
[INFO] [launch]: All log files can be found below /home/sickle/.ros/log/2026-02-12-13-47-18-116129-ROSbuntu-91834
[INFO] [launch]: Default logging verbosity is set to INFO
namespace:        
transport:        udp4
middleware:       dds
verbose:          4
discovery:        7400
port:             2019
command:          arducopter
model:            quad
speedup:          1
slave:            0
sim_address:      127.0.0.1
instance:         0
defaults:         /home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/copter.parm,/home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/dds_udp.parm
synthetic_clock:  True
command:          mavproxy.py
master:           tcp:127.0.0.1:5760
sitl:             127.0.0.1:5501
out:              127.0.0.1:14550
console:          False
map:              False
[INFO] [micro_ros_agent-1]: process started with pid [91835]
[INFO] [dds_udp.parm --synthetic-clock -2]: process started with pid [91837]
[INFO] [mavproxy.py -3]: process started with pid [91839]
[dds_udp.parm --synthetic-clock -2] Setting SIM_SPEEDUP=1.000000
[dds_udp.parm --synthetic-clock -2] Ignoring stale command-line parameter '-S'Suggested EK3_DRAG_BCOEF_* = 17.209, EK3_DRAG_MCOEF = 0.209
[dds_udp.parm --synthetic-clock -2] Starting sketch 'ArduCopter'
[dds_udp.parm --synthetic-clock -2] Starting SITL input
[dds_udp.parm --synthetic-clock -2] Using Irlock at port : 9005
[dds_udp.parm --synthetic-clock -2] Waiting for connection ....
[dds_udp.parm --synthetic-clock -2] bind port 5760 for SERIAL0
[dds_udp.parm --synthetic-clock -2] SERIAL0 on TCP port 5760
[micro_ros_agent-1] [1770884238.163752] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 2019
[micro_ros_agent-1] [1770884238.164102] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[dds_udp.parm --synthetic-clock -2] Connection on serial port 5760
[dds_udp.parm --synthetic-clock -2] Loaded defaults from /home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/copter.parm,/home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/dds_udp.parm
[dds_udp.parm --synthetic-clock -2] bind port 5762 for SERIAL1
[dds_udp.parm --synthetic-clock -2] SERIAL1 on TCP port 5762
[dds_udp.parm --synthetic-clock -2] bind port 5763 for SERIAL2
[dds_udp.parm --synthetic-clock -2] SERIAL2 on TCP port 5763
[dds_udp.parm --synthetic-clock -2] Loaded defaults from /home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/copter.parm,/home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/dds_udp.parm
[dds_udp.parm --synthetic-clock -2] Home: -35.363262 149.165237 alt=584.000000m hdg=353.000000
[dds_udp.parm --synthetic-clock -2] Smoothing reset at 0.001
[dds_udp.parm --synthetic-clock -2] validate_structures:538: Validating structures
[dds_udp.parm --synthetic-clock -2] Waiting for internal clock bits to be set (current=0x00)
[dds_udp.parm --synthetic-clock -2] Loaded defaults from /home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/copter.parm,/home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/dds_udp.parm
[dds_udp.parm --synthetic-clock -2] Loaded defaults from /home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/copter.parm,/home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/dds_udp.parm
[dds_udp.parm --synthetic-clock -2] Loaded defaults from /home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/copter.parm,/home/sickle/ardu_ws/install/ardupilot_sitl/share/ardupilot_sitl/config/default_params/dds_udp.parm
[... 135 lines cut ...]
[mavproxy.py -3] Received 1378 parameters
[mavproxy.py -3] Saved 1369 parameters to mav.parm
[mavproxy.py -3] AP: EKF3 IMU0 initialised
[mavproxy.py -3] AP: EKF3 IMU1 initialised
[mavproxy.py -3] AP: AHRS: EKF3 active
[mavproxy.py -3] AP: DDS: No ping response, disconnecting
[mavproxy.py -3] AP: EKF3 IMU1 tilt alignment complete
[mavproxy.py -3] AP: EKF3 IMU0 tilt alignment complete
[mavproxy.py -3] AP: EKF3 IMU1 MAG0 initial yaw alignment complete
[mavproxy.py -3] AP: EKF3 IMU0 MAG0 initial yaw alignment complete
[mavproxy.py -3] AP: GPS 1: probing for u-blox at 230400 baud
[mavproxy.py -3] AP: GPS 1: detected u-blox
[mavproxy.py -3] AP: DDS: No ping response, exiting
[mavproxy.py -3] fence present
[mavproxy.py -3] Flight battery 100 percent
[mavproxy.py -3] AP: EKF3 IMU1 origin set
[mavproxy.py -3] AP: Field Elevation Set: 584m
[mavproxy.py -3] AP: EKF3 IMU0 origin set
[mavproxy.py -3] AP: DDS: No ping response, exiting
[mavproxy.py -3] AP: DDS: No ping response, exiting
[mavproxy.py -3] AP: EKF3 IMU0 is using GPS
[mavproxy.py -3] AP: EKF3 IMU1 is using GPS
[mavproxy.py -3] AP: DDS: No ping response, exiting
[mavproxy.py -3] AP: DDS: No ping response, exiting
[mavproxy.py -3] AP: DDS: No ping response, exiting