ArduPilot/ardupilot #30855: fatal error: mavlink/v2.0/common/mavlink.h

hgf1989, 2025-08-07:
About mavlink ，allway cant find 
In file included from ../../libraries/GCS_MAVLink/GCS.h:7,
                 from ../../libraries/APM_Control/AP_PitchController.cpp:23:
../../libraries/GCS_MAVLink/mavlink_auth.h:6:10: fatal error: mavlink/v2.0/common/mavlink.h: No such file or directory
    6 | #include "mavlink/v2.0/common/mavlink.h"

we can find   no mavlink in the gcs ，

## Bug report

**Issue details**

_Please describe the problem_

**Version**
rover 4.6. 2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ ✔] Rover
[  ] Submarine

**Airframe type**
rover

**Hardware type**
omnibusf4 

**Logs**
In file included from ../../libraries/GCS_MAVLink/GCS.h:13,
                 from ../../libraries/GCS_MAVLink/GCS_Common.cpp:22:
../../libraries/GCS_MAVLink/GCS_MAVLink.h:15: warning: "MAVLINK_START_UART_SEND" redefined
   15 | #define MAVLINK_START_UART_SEND(chan, size) comm_send_lock(chan, size)
      |
In file included from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\common.h:29,
                 from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\mavlink.h:32,
                 from ../../libraries/GCS_MAVLink/mavlink_auth.h:6,
                 from ../../libraries/GCS_MAVLink/GCS_Common.cpp:17:
g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\protocol.h:29: note: this is the location of the previous definition
   29 | #define MAVLINK_START_UART_SEND(chan, length)
      |
In file included from ../../libraries/GCS_MAVLink/GCS.h:13,
                 from ../../libraries/GCS_MAVLink/GCS_Common.cpp:22:
../../libraries/GCS_MAVLink/GCS_MAVLink.h:16: warning: "MAVLINK_END_UART_SEND" redefined
   16 | #define MAVLINK_END_UART_SEND(chan, size) comm_send_unlock(chan)
      |
In file included from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\common.h:29,
                 from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\mavlink.h:32,
                 from ../../libraries/GCS_MAVLink/mavlink_auth.h:6,
                 from ../../libraries/GCS_MAVLink/GCS_Common.cpp:17:
g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\protocol.h:33: note: this is the location of the previous definition
   33 | #define MAVLINK_END_UART_SEND(chan, length)
      |
In file included from ../../libraries/GCS_MAVLink/GCS.h:13,
                 from ../../libraries/GCS_MAVLink/GCS_Common.cpp:22:
../../libraries/GCS_MAVLink/GCS_MAVLink.h:23: warning: "MAVLINK_COMM_NUM_BUFFERS" redefined
   23 | #define MAVLINK_COMM_NUM_BUFFERS 5
      |
In file included from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\protocol.h:4,
                 from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\common.h:29,
                 from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\mavlink.h:32,
                 from ../../libraries/GCS_MAVLink/mavlink_auth.h:6,
                 from ../../libraries/GCS_MAVLink/GCS_Common.cpp:17:
g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\mavlink_types.h:182: note: this is the location of the previous definition
  182 | # define MAVLINK_COMM_NUM_BUFFERS 4
      |
In file included from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\common.h:29,
                 from g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\common\mavlink.h:32,
                 from ../../libraries/GCS_MAVLink/mavlink_auth.h:6,
                 from ../../libraries/GCS_MAVLink/GCS_Common.cpp:17:
g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\protocol.h: In function 'uint16_t _MAV_RETURN_uint16_t(const mavlink_message_t*, uint8_t)':
g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\protocol.h:273:11: error: cast from 'const char*' to 'const uint16_t*' {aka 'const short unsigned int*'} increases required alignment of target type [-Werror=cast-align]
  273 | { return *(const TYPE *)(&_MAV_PAYLOAD(msg)[ofs]);}
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g:\ardupilot\build\omnibusf4\libraries\gcs_mavlink\include\mavlink\v2.0\protocol.h:275:1: note: in expansion of macro '_MAV_MSG_RETURN_TYPE'
  275 | _MAV_MSG_RETURN_TYPE(uint16_t)
      | ^~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus.exe: some warnings being treated as errors

IamPete1, 2025-08-08:
Don't forget to update your submodules. https://ardupilot.org/dev/docs/git-submodules.html

peterbarker, 2025-10-19:
No answer, I'm going to assume the problem was resolved.

closed 2025-10-19 by peterbarker