ArduPilot/ardupilot #30886: Tools: clarify build_options.py's comments around dependency table

rmackay9, 2025-08-12:
It would be good to clarify the meaning of each column of the [build_options.py script's dependency table](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/build_options.py#L31).

I struggled with this for [this PR](https://github.com/ArduPilot/ardupilot/pull/30718) and maybe I'm slow but it wasn't clear to me what each column was for.
```

# list of build options to offer NOTE: the dependencies must be
# written as a single string with commas and no spaces,
# eg. 'dependency1,dependency2'
BUILD_OPTIONS = [
    Feature('AHRS', 'EKF3', 'HAL_NAVEKF3_AVAILABLE', 'Enable EKF3', 1, None),
    Feature('AHRS', 'EKF2', 'HAL_NAVEKF2_AVAILABLE', 'Enable EKF2', 0, None),
```

bhavyasree-A8, 2025-12-26:
Hi, i’m new to ArduPilot and was going through the build system docs.
I found this issue while trying to understand build_options.py . If this is still open and no one is actively working on it, I’d be happy to help by clarifying the comments around the dependency table. Please let me know if that would be useful.

bhavyasree-A8, 2025-12-27:
Thank you . I’ll take a look and work on clarifying the comments. I’ll ask if I have any questions.

closed 2025-12-30 by peterbarker