ArduPilot/ardupilot #30920: BLHeli passthrough on Blitz F745 broken by PR #29590

ethomas997, 2025-08-17:
## Bug report

**Issue details**

BLHeli passthrough on an iFlight Blitz F745 Mini (connected to a Speedybee Mini BLS V2 8Bit 35A 20x20 4in1 ESC) stops working when code from PR #29590 ("The Mother of All BLHeli Passthrough Fixes") is in use.  ESC Configurator can connect, but reading ESC settings fails (after a delay); same with BLHeli Suite.

**Version**
Copter 4.6.2 (stable) and 4.7.0 (dev) show the issue.  Reverting to Copter 4.6.1 fixes it.

**Platform**
Copter

**Hardware type**
iFlight Blitz F745 Mini
Speedybee Mini BLS V2 8Bit 35A 20x20 4in1 ESC

**Additional Info**
I've done testing via builds on the main branch; at the last commit before the PR (https://github.com/ArduPilot/ardupilot/commit/7cfb84e30d5aeb263c48167341240ffde242f57a) the passthrough works fine; at first commit containing the PR (https://github.com/ArduPilot/ardupilot/commit/ba1a02b386754517da486dfa4db8275999b0d784) the passthrough stops working.  I haven't looked into the code details, but let me know if I can help with testing code changes or if there's something you'd like me to try.

ethomas997, 2025-08-31:
@andyp1per Any thoughts on this one?

seankovacs, 2025-09-19:
Wanted to confirm the same. 4.6.1 works fine, 4.6.2, I can only read 2 ESCs even after successive tries. Currently using Bluejay.

andyp1per, 2025-09-19:
Should be fixed by #31077

ethomas997, 2025-09-22:
I tested on a Copter build of v4.6.2 with these changes applied -- works, looks good!